microsoft/vscode-arduino #354: Feature Request: Better Build Manager

rei-vilo, 2017-07-03:
As at today, the Arduino extension for Visual Studio Code relies on the `arduino_builder` as build manager. Problem is, it is especially slow.

The feature request is about using the standard GNU `make` with makefiles as build manager.

rei-vilo, 2017-07-03:
See this [post](https://gitter.im/Microsoft/vscode-arduino?at=593f5b91ca6d4ae80c0a53c8) in Gitter

> Rome Li @akaroml June 13 05:27
>@sytone PlatformIO has its own make logic instead of using what comes with Arduino IDE. That's why it can accept extra build parameters. We'll investigate how to achieve that goal but still keep the vanilla Arduino tool chain.

EricCDI13, 2017-07-06:
Agree
but why **upload** is really slower than with the Arduino IDE ? 
Not the same builder used ?

yaohaizh, 2017-07-06:
Currently, the upload of VSCode is from this comands from this page:  
https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc
which doesn't reuse the built out binaries. 
It  can use another tool calls [arduino-builder](https://github.com/arduino/arduino-builder) but it take times.

SeanChromech, 2017-08-18:
Not using cached binaries seems to really increase compile times, especially when using large libraries.  The official Arduino IDE does this, but VS Code seems to do a full rebuild on every verify.  Adding --preserve-temp-files doesn't help, probably unsurprisingly.

liydu, 2018-02-09:
Wait to see what **arduino-cli** can provide.

rei-vilo, 2018-02-09:
@liydu, Sorry to disagree. The [post](https://gitter.im/Microsoft/vscode-arduino?at=593f5b91ca6d4ae80c0a53c8) in Gitter was about exploring plain `make` along the Arduino tools solution. 

Otherwise, by over-relying on Arduino tools, the whole Visual Code Studio Arduino extension is a mere front-end, not a real solution.

czgtest, 2018-03-06:
@ArthurMa1978 , this issue is duplicate with issue #478 ,  Can we close this issue?

rei-vilo, 2018-03-06:
I’m not sure this is a duplicate. 

+ Issue #478 is about implementing incremental build from the Arduino builder. 
+ This issue #354 is about implementing an autonomous build that no longer relies on the Arduino builder. Recommended solution is to use `make` with makefiles.

ArthurMa1978, 2018-03-06:
@rei-vilo agree, it's not the same issue.
And for `make`, as liya mentioned above, we are working on a new iteration of this extension, both `make` and `arduino-cli` are candidates.

rei-vilo, 2018-03-07:
@czgtest Maybe you can help with **How to pass VSC variables from c_cpp_properties.json to a Task?** #742.

rei-vilo, 2019-08-15:
Any news on this and related topics?

robotdad, 2020-10-30:
This extension relies on the integration with Arduino tooling. It was designed as straight integration for existing Arduino projects. If you use a different build tool like make or cmake many of the Arduino concepts here won't map in, e.g. board manager, libraries, etc. won't integrate with updating the other build script. I am open to further discussion, but I don't think this is something we would pursue. If you want to leverage Arduino tools for use from say CMake I think a toolchain file that knows where the Arduino compilers, libs, etc. are would be a good approach. It leaves open questions about upload etc. though. So which pieces should work together and which not?

rei-vilo, 2020-10-30:
With Arduino developing the [Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide) based on Eclipse Theia, the Visual Studio Code extension for Arduino might no longer be relevant.

closed 2020-10-30 by rei-vilo

robotdad, 2020-10-30:
I think your feedback is valid, it raises some interesting questions so thank you for it.

rei-vilo, 2020-10-30:
On the make topic, I'm using [embedXcode](https://embedxcode.weebly.com).