microsoft/vscode-arduino #355: Feature Request: Support for Other IDEs

rei-vilo, 2017-07-03:
Although the Arduino IDE supports many boards thanks to the boards packages, some popular boards like Teensy / Teensyduino and Launchpad / Energia are not supported through the Boards Manager because of Arduino incompatibilities. 

For example, Launchpad / Energia has committed a pull request, [Add build.project_path to global properties #119](https://github.com/arduino/arduino-builder/pull/119), but ArduinoCC hasn't answered since February 2016.

techniq, 2017-08-19:
I would love to see Teensy supported.

hsiboy, 2018-01-24:
+1 for Teensy support, although this may be tricky as currently Teensyduino patches the underlying Arduino (IDE)  jars.

yaohaizh, 2018-01-25:
This extension is targeting official Arduino IDE compatible board support and it requires Arduino IDE support. So as long the board can be loaded into Arduino IDE via additional URL, we can support the board. Otherwise, it is not the scope of this extension.

atestani, 2018-01-29:
There is Arduino IDE board support for the Teensy boards that get installed with Teensyduino supplied by PJRC. You first install the Arduino IDE and then Teensyduino.   It is located at <arduino IDE path>/hardware/teensy and there is a board.txt and platform.txt file there plus the core and libraries.  It is right alongside he Arduino equivalent.

Except for the proprietary uploader and dedicated bootloader chip on the Teensy boards there is no difference to Arduino.  One can create a new "board" in the teensy/board.txt file and use any other uploader (e.g. JTAG or SWD) (assuming wires are added to the stock Teensy boards for SWD interface.  Later version Teensy boards (e.g. 3.5 and 3.6) have pads for adding such wires

The VS Code Arduino plugin should be able to support the Teensy boards which provide powerful Kinetis ARM processors.  Would you please consider adding this to your great plug-in!?    

Thanks
There is Arduino IDE board support for the Teensy boards that get installed with Teensyduino supplied by PJRC. YOu first install the Arduino IDE and then Teensyduino.   It is located at <arduino IDE path>/hardware/teensy and there is a board.txt and platform.txt file there plus the core and libraries.  

Except for the proprietary uploader and dedicated bootloader chip on the Teensy boards there is no difference to Arduino.  One can create a new board in the teensy/board.txt file and use any other uploader (e.g. JTAG or SWD) but not from within the Arduino environment (although there must be a way to do this but i haven't gone that far).

Net here is that the VS Code Arduino plugin should be able to support the Teensy boards which provide powerful Kinetis ARM processors.  

Would you please consider adding this as further function in your great plug-in!?    

Thanks

czgtest, 2018-03-06:
@rei-vilo  & @atestani   I have not used Teensy as before, can you provide some more detailed about the problems you want us to fix and which feature you want us to add ?  Thanks a lot!

rei-vilo, 2018-03-06:
Both Teensy and LaunchPad, albeit complying with the Arduino specifications, require some modifications of the original Arduino IDE. 

For Teensy, Paul Stoffregen explains [on this post](https://forum.pjrc.com/threads/28488-Arduino-Board-Manager?p=73762&viewfull=1#post73762) the technical reasons for a separate IDE. 

On the case of Energia, the pull request [Add build.project_path to global properties #119](https://github.com/arduino/arduino-builder/pull/119) is ready since February 2016, but Arduino hasn't answered yet —and two years after, will it consider it? Here the clear opportunity for Visual Studio Code.

czgtest, 2018-03-07:
@rei-vilo this request is very valuable, we will sync with team-mate for it. Thanks a lot.

rei-vilo, 2018-03-07:
For Energia, all the source code is available at https://github.com/energia.

rei-vilo, 2018-03-08:
> This extension is Arduino only, so I’d not like to support them even there has some workarounds to let these boards playing with Arduino IDE. But we’re working on another extension, maybe that extension can support these boards.

Arthur:

Thank you for your prompt answer.

Is there a repository for the other extension?

Best regards,

— Rei Vilo
rei_vilo@hotmail.com<mailto:rei_vilo@hotmail.com>

JamesCoyle, 2018-07-28:
@ArthurMa1978 Adding Teensy support is super easy. I've got it working by appending the `boards.txt` and `platform.txt` files found in `C:\Program Files (x86)\Arduino\hardware\teensy\avr` to those found in `C:\Program Files (x86)\Arduino\hardware\arduino\avr`. Then you simply copy the contents of the `cores` folder from the teensy folder to the arduino folder. 

Can't this extension just be made aware of the teensy folder and pull files directly from there?

klyemar, 2018-07-28:
@JamesCoyle, I got the boards to be recognized by appending the contents of boards.txt but I'm curious how you went about copying the contents from platform.txt. For instance, I see that some settings like compiler.path will presumably overwrite the existing values, which I guess *should* be fine. 

I appended the contents to the bottom and am currently getting "fatal error: Arduino.h: No such file or directory" at compile time. I'm not sure if this is directly related, or if it's a result of my trying to port our existing code from an Arduino to the Teensy.

rei-vilo, 2019-08-15:
@JamesCoyle

Does upload works as expected? Teensy has specific upload and terminal procedures, hence the dedicated IDE Teensyduino.

rei-vilo, 2019-08-15:
@yaohaizh

> This extension is targeting official Arduino IDE compatible board support and it requires Arduino IDE support. So as long the board can be loaded into Arduino IDE via additional URL, we can support the board. Otherwise, it is not the scope of this extension.

By providing documentation (especially on hard-coded elements) and guidance on how to adapt the VSCode Arduino extension to other Arduino-compliant platforms, VSCode could position itself as the next IDE of choice for fast prototyping.

sarpik, 2019-10-09:
Hello, any updates on this? 
Specifically, teensy support?

denkyuu, 2019-11-08:
Just adding another +1 for teensy support. The latest mac update broke teensyduino for the Arduino IDE and I'm looking for an alternative. I use VS code for everything else, so having my teensy development wrapped up into the same environment would be incredible.

atestani, 2019-11-09:
Are you using PlatformIO which is an extension for VSCode?  It has support for Teensy boards as well as custom boards with the Kinetis MCUs as in Teensy.  PlatformIO has support for over 700 boards and has a full function debugger plus _many_ other features.  

My post in this topic of Jan 29, 2018 was aimed at PlatformIO so I don't know why it ended up here.

PlatformIO is the best thing to come to embedded development ever!  Check it out...

sarpik, 2019-11-09:
Yo @atestani I just found out about platformio (both in vscode and it's cli) and it's super awesome, thank you!

ackvf, 2020-02-11:
There seem to be a solution https://github.com/luni64/VisualTeensy

taken from this discussion https://forum.pjrc.com/threads/48878-How-to-add-Teensy-board-to-VS-Code-with-Arduino-plugin

rei-vilo, 2020-02-11:
Alas, only for Windows.