microsoft/vscode-arduino #356: Feature Request: Better GUI

rei-vilo, 2017-07-03:
The GUI of Visual Studio Code is pretty limited. 

Is there a way to implement  

* a dedicated **Arduino** menu, or 
* a floating tool-bar with some buttons for **Build** / **Verify** and **Upload** / **Run**?

testforstephen, 2017-07-04:
@rei-vilo yes, VSCode have strict limit on GUI layout. it seems we could add some buttons on editor action bar. something like this
[image]

czgtest, 2018-03-06:
Good request, we will investigate how to improve the GUI.

rei-vilo, 2019-08-15:
A temporary solution is the **Shortcut Menu Bar** [jerrygoyal.shortcut-menu-bar](https://marketplace.visualstudio.com/items?itemName=jerrygoyal.shortcut-menu-bar).

robotdad, 2020-10-30:
There was a verify and upload button added since this was filed. There are also newer options down in the bottom status bar. I'm closing based on age and that some of these UI suggestions have been taken. Please open a new issue with specific suggestions off the current release if you have additional feedback.

closed 2020-10-30 by robotdad

rei-vilo, 2020-10-30:
See 

+ [Add some API to support GUI customization](https://github.com/Microsoft/vscode/issues/1833) and 
+ [Add an optional configurable toolbar below the menu](https://github.com/microsoft/vscode/issues/41309)

robotdad, 2020-10-30:
Thank you for those specific pointers.

MikeRosoft, 2020-10-30:
And... ? "Thank you for those specific pointers; as always, we're going to summarily ignore them, and summarily close any further requests pointing in the same direction"?