microsoft/vscode-arduino #357: Bug: Can't Find .elf File When Debugging

rei-vilo, 2017-07-03:
When trying to debug, Visual Studio Code can't find the `.elf` file and displays the error

> ERROR Cannot find the elf file.

although the `.elf` is available under the `..build` folder.

[image]

rei-vilo, 2017-07-03:
Same issue for Arduino extension `0.2.4` on Visual Studio Code `Version 1.13.1 (1.13.1)`.

yaohaizh, 2017-07-04:
@rei-vilo: Could you show the launch.json file context?

andxu, 2017-07-04:
Is your arduino.json configuration like this:
{
    "port": "/dev/cu.usbmodem1412",
    "board": "arduino:samd:mzero_pro_bl_dbg",
    "sketch": "LoadLibrary.cpp"
}

you can delete the sketch line or change it into embed1.ino ,the sketch is supposed to be the ino file.

rei-vilo, 2017-07-04:
Please find the two files:

* **arduino.json**, consistent with the sketch `embed1.ino`

```
{
    "board": "arduino:samd:mzero_pro_bl_dbg",
    "sketch": "embed1.ino",
    "port": "/dev/cu.usbmodem1422"
}
```

* **launch.json**

```
{
    "version": "0.2.0",
    "configurations": [
    
        {
            "name": "Arduino",
            "type": "arduino",
            "request": "launch",
            "program": "${file}",
            "cwd": "${workspaceRoot}",
            "MIMode": "gdb",
            "targetArchitecture": "arm",
            "miDebuggerPath": "",
            "debugServerPath": "",
            "debugServerArgs": "",
            "customLaunchSetupCommands": [
                {
                    "text": "target remote localhost:3333"
                },
                {
                    "text": "file ${file}"
                },
                {
                    "text": "load"
                },
                {
                    "text": "monitor reset halt"
                },
                {
                    "text": "monitor reset init"
                }
            ],
            "stopAtEntry": true,
            "serverStarted": "Info\\ :\\ [\\w\\d\\.]*:\\ hardware",
            "launchCompleteCommand": "exec-continue",
            "filterStderr": true,
            "args": []
        }
    ]
}
```

rei-vilo, 2017-07-04:
Now, if I change `"sketch": "embed1.ino",` for `"sketch": "LocalLibrary.cpp",` in **arduino.json**, debugging works.

It looks like Visual Code Studio takes the wrong name for the target, `LocalLibrary.cpp.elf` instead of `embed1.ino`. 

Now, the name of the sketch `embed1.ino` differs from the name of the folder `vscArduino`. The Arduino IDE checks that consistency, so should Visual Code Studio.

andxu, 2017-07-04:
1. The arduino extension will always search the workspace for *.ino and uses the first file name as the expected elf file name if sketch is not set, so you need not to set the sketch at all, the debugger will use the ino in your project.
2. Arduino IDE verifies the folder name against the ino filename and it doesn't verify it with its command line, so arduino extension will just use the command line version of Arduino IDE, and arduino extension will allow you to have different filename of ino from folder name.

closed 2017-07-04 by rei-vilo