microsoft/vscode-arduino #358: Bugfix/Tip for Install-package arduino error

hoan-cosmos, 2017-07-05:
Using Windows 7 Ultimate 64 bit, Visual Studio Code 1.13.1, Arduino extension 0.2.4 and latest jdk and jre.

Problem encountered when downloading package from Arduino Package Manager (full internet connection).

_**Output**:_ 
[Starting] Install package - arduino...
Loading configuration...
Initializing packages...
Preparing boards...
Downloading platforms index... 
Using proxy DIRECT
java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:113)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:330)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:305)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:283)
	at processing.app.Base.<init>(Base.java:349)
	at processing.app.Base.guardedMain(Base.java:222)
	at processing.app.Base.main(Base.java:136)
Caused by: java.io.IOException: Received invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:188)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:123)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:111)
	... 6 more
[Error] Exit with code=255

**My SOLUTION**: After I deleted C:\Users\your_computer_name\AppData\Local\Arduino15\package_index.json and package_index.json.tmp, the download started normally.

Reason for this may due to unstability of internet connection during unsucessful download in the past, that may lead to error in these 2 files.

Hope to help someone with the same problem.

`

liydu, 2018-02-09:
Wait to see what **arduino-cli** can provide to enhance download interruption issue.

czgtest, 2018-03-07:
@hoan-cosmos  have you got this issue solved?

hoan-cosmos, 2018-04-21:
@czgtest Yes, I solved this and posted the solution above.

closed 2018-05-21 by czgtest