microsoft/vscode-arduino #359: Support Over-the-Air (OTA) uploads

sernaleon, 2017-07-05:
Currently some boards with built-in wifi (ESP8266, ESP32) support uploads over the air. This feaure is available in other VSCode extensions such as [PlatformIO](http://docs.platformio.org/en/latest/projectconf/section_env_upload.html?highlight=OTA).

My suggestion would be to be able to select an IP address instead of a serial port when the board supports OTA.

leppie, 2017-07-05:
You can just set `port` to an IP address in `arduino.json`.

sernaleon, 2017-07-06:
Thanks @leppie, it works indeed! But at the cost of disabling the serial monitor, since this one will try to open the communication with the provided IP (instead of an actual serial port).

So in short, the raw functionality is there, but not via commands and conflicting with the serial monitor.

leppie, 2017-07-06:
You can just keep something else open for serial comms, like Termite. That's what I do :D

sernaleon, 2017-07-08:
I created a pull request with my suggested solution. You can find it here: https://github.com/Microsoft/vscode-arduino/pull/362

tommitytom, 2017-10-11:
leppie's suggestion does not work for me, I get the following error:
`avrdude: ser_open(): can't open device "192.168.1.235": No such file or directory`

Am I missing something?

Robotto, 2017-11-11:
Adding the IP address as port number does not work when OTA is set up with access restriction.. i.e. an OTA flash that requires a password.

czgtest, 2018-03-06:
this issue is dulplicate with #362

czgtest, 2018-03-07:
the issue #362 , #359 has the same request.

czgtest, 2018-03-07:
#452  and #454

izamoramx, 2021-01-04:
> Adding the IP address as port number does not work when OTA is set up with access restriction.. i.e. an OTA flash that requires a password.

Is there a solution for this issue? I am currently dealing with the same situation

Invisibleman1002, 2022-05-02:
Was searching for another issue and ran into this thread.
I thought I would share what I did for my esp8266 for doing OTA upload.

I created a tasks.json file using the data below..
This allows me to use OTA, with Serial Monitor running.  

The second task allows me to send data to the com port again after I realized I had it opened already in YAT.  
Instead of waiting for it to compile again, I just press F1 > RUN TASK > COM UPLOAD.
Of course, I still have to remember to disconnect from the com port before uploading.
OTA works great but make sure your paths are correct, as well as IP address and password.

**Again, this was used for esp8266, path will be different for esp32**
  
```
{
  // See https://go.microsoft.com/fwlink/?LinkId=733558
  // for the documentation about the tasks.json format
  "version": "2.0.0",
  "tasks": [
    {
      "label": "OTA Upload",
      "type": "shell",
      "command": "py",
      "args": [
        "%LOCALAPPDATA%\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2\\tools\\espota.py",
        "-i",
        "192.168.0.232",
        "-p",
        "8266",
        "-a",
        "admin",
        "-f",
        "${workspaceFolder}\\..\\ArduinoOutput\\${fileBasenameNoExtension}\\${fileBasename}.bin"
      ],
      "options": {
        "shell": {
          "executable": "cmd.exe",
          "args": ["/c"]
        }
      }
    },
    {
      "label": "COM Upload",
      "type": "shell",
      "command": "py",
      "args": [
        "%LOCALAPPDATA%\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2\\tools\\esptool\\esptool.py",
        "--chip",
        "esp8266",
        "--port",
        "COM20",
        "--baud",
        "921600",
        "write_flash",
        "--flash_mode",
        "dio",
        "--flash_size",
        "detect",
        "0x0",
        "${workspaceFolder}\\..\\ArduinoOutput\\${fileBasenameNoExtension}\\${fileBasename}.bin"
      ],
      "options": {
        "shell": {
          "executable": "cmd.exe",
          "args": ["/c"]
        }
      },
      "problemMatcher": []
    }
  ]
}
```
[image]
Can also grab the task.json from here:
https://github.com/Invisibleman1002/dynaHTML/blob/master/.vscode/tasks.json

JumpMaster, 2023-02-18:
When using ArduinoOTA with the new built-in Arduino-CLI I receive the following error on upload.  I have the port variable in arduino.jso set to the IP address of the ESP32.  No other authentication required.

Does anyone else have this working?

[Error] Uploading sketch 'sketch.ino': Exit with code=7

mark-harding, 2023-08-29:
vscode ota only works with ide 1.8