microsoft/vscode-arduino #360: How to Play Sound using IoT Devkit (AZ3166)?

dwzy, 2017-07-07:
Hi Guys,

I have an AZ3166 Devkit, and have a question as the subject.
I read from the Doc that the Audio.h class has no Play method, but I notice that there are "Headphone" and "Speaker" labels on the kit.

Thanks!
Derek

liydu, 2017-07-08:
Hi @dwzy , we currently have not yet added Arduino API for play. But it will come very soon. The "Speaker" label on the board will be obsolete as it's just two golden pins there for testing purpose.

You can also report feedback and problems on using DevKit in our Gitter channel: https://gitter.im/Microsoft/azure-iot-developer-kit

dwzy, 2017-07-08:
Good to hear API for Play is coming very soon!
Thank you Liya!

BTW, will Ardunio support C# or Python in the future?

Regards,
Derek

liydu, 2017-07-09:
Arduino itself is based on C / C++ as its foundation. So the odds to support other languages are not big.  There are some implementations here like Makecode.com on Micro:bit and Micropython. But for DevKit specifically, we are open to support high level languages as long as our developers are asking for it.

dwzy, 2017-07-09:
Thank you very much Liya!
I am working at Microsoft China and my role is ATS. Good to know you!
I do hope DevKit can support C# and Python.

Cheers,
Derek

closed 2017-07-13 by yaohaizh