microsoft/vscode-arduino #363: Problem to Upload code

agaelema, 2017-07-08:
I'm trying to use the VS Code but there is a problem to upload the code.

[Starting] Upload sketch - Blink.ino
Carregando configura��o...
java.lang.NullPointerException
	at processing.app.BaseNoGui.selectSerialPort(BaseNoGui.java:1088)
	at processing.app.helpers.CommandlineParser.parseArgumentsPhase1(CommandlineParser.java:143)
	at processing.app.Base.<init>(Base.java:273)
	at processing.app.Base.guardedMain(Base.java:219)
	at processing.app.Base.main(Base.java:132)
[Error] Exit with code=255

The port and board are correctly selected

Eskibear, 2017-07-08:
@agaelema I think it would be more helpful to share your board type and other details.

agaelema, 2017-07-08:
@Eskibear , I tested with two boards (Nano and Uno), and the result was the same. The serial does not port appear to be problem, because it's possible to monitoring it port when the serial monitor is enabled.
[image]

yaohaizh, 2017-07-09:
@agaelema  What's the version of the Arduino IDE that you used here?

jeidon, 2017-07-09:
I experienced the same issue using Arduino 1.6.12. Didn't realize I was so out of date. Updated to 1.8.3 and it worked.

agaelema, 2017-07-09:
Exactly, I'm using version 1.6.12.

By the prerequisites I thought it would work without problems (I had no plans to update the version).

agaelema, 2017-07-09:
VS Code works well changing the Arduino IDE to 1.8.3

closed 2017-07-10 by yaohaizh