microsoft/vscode-arduino #366: include file not found in browse.path.

agaelema, 2017-07-10:
The VS Code is showing some problems in sketches. The code compile without any problem, but is not possible to view the content of include files (clicking in the function with "Ctrl" pressed).

Is there a way to resolve this?

Arduino IDE 1.8.3, VS Code 1.13.1.

Thanks

[image]

yaohaizh, 2017-07-11:
@agaelema  For this issue, it is related to the VSCode C++ extension.  You can find the details about how to resolve this issue: 
https://github.com/Microsoft/vscode-cpptools/blob/master/Documentation/LanguageServer/FAQ.md#why-do-i-have-red-underlines-everywhere-after-updating-to-the-latest-version

https://github.com/Microsoft/vscode-cpptools/blob/master/Documentation/LanguageServer/FAQ.md#what-is-the-difference-between-includepath-and-browsepath-in-c_cpp_propertiesjson

svq98, 2017-07-11:
That does the trick. 
Is there a way to set it as a default property for all projects. Every time I open a new project I have to copy all the settings to the new project.

PeterSek, 2017-07-12:
Had same problem, I am using Arduino extensions and resolved by adding browse path as show below (i use custom libs). It fixed the problem...Is this correct way to resolve this?

[image]

mattywashington, 2017-10-24:
svq98,

I was able to set the default property file by doing the following.  

1. As the document provided above instructs, set the C_Cpp.intellienseengine to "Tag Parser"

2. Locate the followin file: .vscode\extensions\vsciot-vscode.vscode-arduino-0.2.9\out\src\arduino\arduino.js

3.  Edit the file by going to line 431, or do a search for "const cppConfig = " and enter in your path like below. 

[image]

That should get it working.

mattywashington, 2017-10-24:
I was able to get my paths as defaults on new projects by doing the following:
- Use Tag Parser as the article above states
- open the arduino.js file located in .vscode\extensions\vsciot-vscode.vscode-arduino-0.2.9\out\src\arduino
- Go to line 215 and edit the section to look like this:
[image]

Also...You can do the same thing for the example sketches by searching for "const cppConfig = " and performing the same steps.

RyanBlace, 2017-10-28:
I fixed it by just ending at 'avr':
[image]

You need everything under AVR.

mattywashington, 2017-11-09:
Your solution won't work to include paths in a default configuration...it also won't add the third party libraries that OP posted that are located in a separate folder.  The last question asked was how to get the include paths, that OP successfully added, set by default so one doesn't have to type them in every time.  

The arduino extension creates the configuration files when you initialize  and it doesn't include a setting option to override what it adds by default, so whatever setting you made in a previous configuration won't follow over.  By default, the arduino extension will only include the hardware libraries of the board you select.  After editing the file in the location that I provided, the include paths that you set will be added by default on the rest of your projects.  The only downside is that every time the extension updates, you have to re-edit the file.  It would be a nice addition if the devs could add an option to set include path settings that can be overridden by the user.