microsoft/vscode-arduino #367: Build tmp directory gets cleared after build

svq98, 2017-07-11:
In the normal arduino IDE, you can verify and it leaves a directory in the C:\Users\mn\AppData\Local\Temp

This directory contains the hex and other cpp files. It can be used to disassemble obj files or use another programmer.

At the end of the build it is cleared. Not very useful!!
Can there be a preference not to clear the build directory?

yaohaizh, 2017-07-11:
@svq98  In your arduino.json file, you can add value of output as your temp folder such as: 
`    "output": ".\\bin"    `

svq98, 2017-07-11:
Thanks for that. In the .vscode directory, there is an arduino.json file and I added that line at the top of the file just after the {
  "output": ".\\bin",

And now it creates a bin directory in the folder of the Arduino sketch directory. 
Thank you very much.

closed 2017-07-11 by svq98