microsoft/vscode-arduino #368: Compile Error 'Serial1 not declared in this scope' on ATMega2560

PeterSek, 2017-07-12:
Hi, getting error on Mega2560 with Serial1

**error: 'Serial1' was not declared in this scope**
   Serial1.begin(115200);            //WiFi
...
**error: 'Serial1' was not declared in this scope**
   if (Serial1.available() > 0) {
       ^
exit status 1

**My Board Config is:**
{
    "board": "arduino:avr:mega",
    "configuration": "cpu=atmega2560",
    "port": "COM4",
    "sketch": "SensorFarmIOWiFi.ino"
}

PeterSek, 2017-07-12:
I must be doing something wrong, Just noted that build.options.json shows arduino:avr:uno but i expect it should be mega2560. what is the best/correct way to fix this?
[image]

yaohaizh, 2017-07-12:
@PeterSek, the vscode arduino extension doesn't use the build.options.json file. Not sure which platform/tools generated this configuration file.  

For this extention, you can set the type in .vscode\arduino.json file or invoke command: "Arduino: Change Board Type" or use the status bar to choose the board type.

PeterSek, 2017-07-12:
thank you @yaohaizh, I tried both but no luck.
FYI yesterday I did my 1st VSC arduino test using 'uno' board 
I noticed that today setting.json shows 'uno' 
I have updated User Settings but these do not include fqbn (no changes in workspace settings)
Tomorrow I might try changing settings.json fqbn (or should this be updated when I choose board as you described?

[image]

yaohaizh, 2017-07-12:
@PeterSek,  our setting are persisted on the arduino.json, the left one of setting.json. Could you show the content of that file?

yaohaizh, 2017-07-12:
@PeterSek, you think you might use another extension.  Our extension is from this link: 
https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino#review-details

PeterSek, 2017-07-12:
@yaohaizh I am using same extension https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino 

here is my arduino.json
[image]

PeterSek, 2017-07-12:
@yaohaizh thank you, I think messed things up, getting different error (output bottom of screen)
Will get back to it tomorrow; P

testforstephen, 2017-07-12:
@PeterSek Looks like you have multiple different arduino extensions installed in your vscode. Note that only these three settings "arduino.path", "arduino.additionalUrls", "arduino.logLevel" are from microsoft arduino extension. Other arduino.xxx settings are not used by our extension. i would suggest you to uninstall other arduino extension first and keep microsoft arduino extension to try again.

PeterSek, 2017-07-12:
Thanks a lot. I will check it out tommorow, will re install vscode if
required
P

PeterSek, 2017-07-13:
Hi @testforstephen yes, you are right, there was a non-Microsoft Arduino 022 also installed
Removing fixed the problem
thank you, closing

closed 2017-07-13 by PeterSek