microsoft/vscode-arduino #370: Selecting Board auto Deselecting

khaled-hamam, 2017-07-13:
Hi,

Whenever I try to select the Board Type, it is deselected automatically once in click on the wanted Board Type, when I opened the arduino.json I found that only one line is written and the replaced when I change any of the Settings (the COM port for example)

khaled-hamam, 2017-07-24:
I've tried many things to solve the problem, the only one that works is by saving the arduino.json as an empty file so it can't save any setting in it.
by this way the problem is solved as it can't save the settings in the JSON file and can't make a new file also, but it keeps getting me errors about the JSON file itself (it's not a big deal as I can upload my code now)

If anyone has another solution that would be great

khaled-hamam, 2017-07-26:
I found what the problem was, I was excluding the .VSCODE folder in the settings, I think I made this while I was trying something when I first tried VS Code and forgot to change in back, now everything is working fine as now it can reach the arduino.json

closed 2017-07-26 by khaled-hamam

saitiku, 2018-01-11:
Can you tell me what you mean by "I found what the problem was, I was excluding the .VSCODE folder in the settings"

vin-ni, 2018-11-03:
where did you find those settings?

desprit, 2019-07-21:
@khaled-hamam 

You saved my day!
I hide things like `.vsode`, virtualenvs and similar stuff from the Explorer to make it more clean. And I have no idea how it may affect Job Board selection. But as soon as I removed `.vscode` from `hide` rules in vscode, I was finally able to select the board type.