microsoft/vscode-arduino #373: "rawModel.packages.forEach is not a function" after executing a command

Mariusz1, 2017-07-17:
If I try to use any command, VSC shows the error message: "rawModel.packages.forEach is not a function". 
It looks like a Java Script problem, maybe some kind of configuration issue? My Arduino is installed in a non-default location but I configured it in settings.
Is there a log file containing more detailed error messages?

yaohaizh, 2017-07-18:
The issue is that the extension found the package_index.json file but failed to parse the content of this package index file. What's the content of your packagee_index.json file?

testforstephen, 2017-07-18:
@Mariusz1 Looks like package_index.json is broken. Please try the workaround. 
Go to directory "C:\Users\Mariusz\AppData\Local\Arduino15", and delete these two files package_index.json and package_index.json.sig.

Mariusz1, 2017-07-18:
Thank you for the responses.
I tried unisntalling the plugin and installing it again. I also deleted the mentioned files (attached below).
[package_index.zip](https://github.com/Microsoft/vscode-arduino/files/1155428/package_index.zip)
After that I started Board manager which executed and downloaded some files. Console:
```
[Starting] Update package index files...
Looking for library 'user32'
Adding paths from jna.library.path: null
Trying user32.dll
Found library 'user32' at user32.dll
Looking for library 'shell32'
Adding paths from jna.library.path: null
Trying shell32.dll
Found library 'shell32' at shell32.dll
Looking for library 'Ole32'
Adding paths from jna.library.path: null
Trying Ole32.dll
Found library 'Ole32' at Ole32.dll
Loading configuration...
Initializing packages...
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Could not find boards.txt in C:\Users\Mariusz\Documents\Arduino\hardware\Arduino_STM32\examples. Is it pre-1.5?
Preparing boards...
Downloading platforms index... 
Using proxy DIRECT
Downloading platforms index... Downloaded 0kb of 187kb.
Downloading platforms index... Downloaded 4kb of 187kb.
Downloading platforms index... Downloaded 8kb of 187kb.
Downloading platforms index... Downloaded 18kb of 187kb.
Downloading platforms index... Downloaded 21kb of 187kb.
Downloading platforms index... Downloaded 31kb of 187kb.
Downloading platforms index... Downloaded 34kb of 187kb.
Downloading platforms index... Downloaded 44kb of 187kb.
Downloading platforms index... Downloaded 47kb of 187kb.
Downloading platforms index... Downloaded 57kb of 187kb.
Downloading platforms index... Downloaded 60kb of 187kb.
Downloading platforms index... Downloaded 63kb of 187kb.
Downloading platforms index... Downloaded 73kb of 187kb.
Downloading platforms index... Downloaded 76kb of 187kb.
Downloading platforms index... Downloaded 79kb of 187kb.
Downloading platforms index... Downloaded 89kb of 187kb.
Downloading platforms index... Downloaded 92kb of 187kb.
Downloading platforms index... Downloaded 103kb of 187kb.
Downloading platforms index... Downloaded 115kb of 187kb.
Downloading platforms index... Downloaded 118kb of 187kb.
Downloading platforms index... Downloaded 120kb of 187kb.
Downloading platforms index... Downloaded 131kb of 187kb.
Downloading platforms index... Downloaded 133kb of 187kb.
Downloading platforms index... Downloaded 145kb of 187kb.
Downloading platforms index... Downloaded 148kb of 187kb.
Downloading platforms index... Downloaded 151kb of 187kb.
Downloading platforms index... Downloaded 156kb of 187kb.
Downloading platforms index... Downloaded 159kb of 187kb.
Downloading platforms index... Downloaded 171kb of 187kb.
Downloading platforms index... Downloaded 174kb of 187kb.
Downloading platforms index... Downloaded 184kb of 187kb.
Downloading platforms index... Downloaded 187kb of 187kb.
Downloading platforms index... 
Using proxy DIRECT
Downloading platforms index... Downloaded 0kb of 0kb.
Downloading platforms index... Downloaded 0kb of 0kb.
Downloading platforms index... 
Using proxy DIRECT
Downloading platforms index... Downloaded 0kb of 26kb.
Downloading platforms index... Downloaded 8kb of 26kb.
Downloading platforms index... Downloaded 10kb of 26kb.
Downloading platforms index... Downloaded 20kb of 26kb.
Downloading platforms index... Downloaded 21kb of 26kb.
Downloading platforms index... Downloaded 23kb of 26kb.
Downloading platforms index... Downloaded 24kb of 26kb.
Downloading platforms index... Downloaded 26kb of 26kb.
Downloading platforms index... 
Using proxy DIRECT
Using proxy DIRECT
Downloading platforms index... Downloaded 0kb of 7kb.
Downloading platforms index... Downloaded 7kb of 7kb.
Downloading platforms index... 
Using proxy DIRECT
Selected board is not available
[Done] Updated package index files.

```
After that the same error is shown. Here are the redownloaded files:
[package_index-redownloaded.zip](https://github.com/Microsoft/vscode-arduino/files/1155441/package_index-redownloaded.zip)

Mariusz1, 2017-08-24:
Any progress on this error? It still shows the same message but the stack trace (console info) is a bit different (e.g. "Promise with no error callback"):

```
rawModel.packages.forEach is not a function: TypeError: rawModel.packages.forEach is not a function
    at BoardManager.loadPackageContent (C:\Users\Mariusz\.vscode\extensions\vsciot-vscode.vscode-arduino-0.2.4\out\src\arduino\boardManager.js:166:27)
    at BoardManager.<anonymous> (C:\Users\Mariusz\.vscode\extensions\vsciot-vscode.vscode-arduino-0.2.4\out\src\arduino\boardManager.js:53:22)
    at Generator.next (<anonymous>)
    at C:\Users\Mariusz\.vscode\extensions\vsciot-vscode.vscode-arduino-0.2.4\out\src\arduino\boardManager.js:9:71
    at __awaiter (C:\Users\Mariusz\.vscode\extensions\vsciot-vscode.vscode-arduino-0.2.4\out\src\arduino\boardManager.js:5:12)
    at BoardManager.loadPackages (C:\Users\Mariusz\.vscode\extensions\vsciot-vscode.vscode-arduino-0.2.4\out\src\arduino\boardManager.js:32:16)
    at ArduinoActivator.<anonymous> (C:\Users\Mariusz\.vscode\extensions\vsciot-vscode.vscode-arduino-0.2.4\out\src\arduinoActivator.js:38:47)
    at Generator.next (<anonymous>)
    at fulfilled (C:\Users\Mariusz\.vscode\extensions\vsciot-vscode.vscode-arduino-0.2.4\out\src\arduinoActivator.js:6:58)
    at process._tickCallback (internal/process/next_tick.js:103:7)

e.onUnexpectedError @ shell.ts:423
(anonymous) @ shell.ts:383
e.onUnexpectedError @ errors.ts:119
o @ errors.ts:139
(anonymous) @ errors.ts:49
(anonymous) @ errors.ts:44

errors.ts:51 WARNING: Promise with no error callback:115
errors.ts:52 Object {exception: null, error: Error: rawModel.packages.forEach is not a function
    at BoardManager.loadPackageContent (C:\Users\…, promise: n.C…s.d…e._oncancel, handler: undefined, id: 115…}
messageService.ts:125 rawModel.packages.forEach is not a function

e.doShow @ messageService.ts:125
e.show @ messageService.ts:104
e.resume @ messageService.ts:63
(anonymous) @ workbench.ts:995
e.invoke @ callbackList.ts:63
e.fire @ event.ts:123
(anonymous) @ quickOpenController.ts:517

```
What can I do to provide more info/investigate it?

yaohaizh, 2017-08-24:
Not get a clue of this issue but add defensive code to avoid this exception: https://github.com/Microsoft/vscode-arduino/pull/400

Mariusz1, 2017-08-24:
Thank you. I assume that the fix will be available in the next version of the plugin. I will let you know about the results :)

yaohaizh, 2017-08-25:
@Mariusz1 The new version was published. You can try this one to see whether the issue repro.

Mariusz1, 2017-08-26:
Thank you - it is fixed now (version 0.2.5) :)

closed 2017-08-26 by Mariusz1