microsoft/vscode-arduino #376: Extension Host hangs the CPU

chrmarti, 2017-07-18:
_From @rstoica on July 13, 2017 17:6_

- VSCode Version: Code 1.14.0 (b8cd5b9556d8b70ea560d35b903422363f6c5c40, 2017-07-10T20:36:40.018Z)
- OS Version: Linux x64 4.4.0-83-generic
- Extensions: 

Extension|Author (truncated)|Version
---|---|---
vscode-markdownlint|Dav|0.9.1
vscode-docker|Pet|0.0.16
gitignore|cod|0.5.0
githistory|don|0.2.1
python|don|0.6.7
cpptools|ms-|0.12.0
csharp|ms-|1.11.0
latex|ms-|0.1.3
vscode-hexdump|sle|1.4.0
cmake|twx|0.0.17
cmake-tools|vec|0.10.1
vscode-arduino|vsc|0.2.4

(2 theme extensions excluded)

---

Steps to Reproduce:

1. Start code normally with all extensions
Effect: after startup and extension load extension host loads the CPU up to 100% and fails to stop even after exiting vscode. The process causing this is under linux:
/usr/share/code/code /usr/share/code/resources/app/out/bootstrap --type=extensionHost
2. To verify this one can check htop - the process above will always be on top.

Reproduces without extensions: No

Please fix this issue, before the June release Code was a dream.

_Copied from original issue: Microsoft/vscode#30639_

chrmarti, 2017-07-18:
Can you retry with the just released 1.14.1?

chrmarti, 2017-07-18:
_From @rstoica on July 13, 2017 23:18_

This patch seems to fix the issue. Thank you!

chrmarti, 2017-07-18:
_From @rstoica on July 14, 2017 0:29_

Hmm, after some time of usage the problem is still there. Absolutely same behavior, just that now it does not start immediately after start-up but with some delay.

chrmarti, 2017-07-18:
Could you try to find the extension that triggers this by disabling them one by one?

yaohaizh, 2017-08-02:
Close this issue as the original issue is closed.

closed 2017-08-02 by yaohaizh