microsoft/vscode-arduino #382: Allow multiple .ino files within single project root

Dismounted, 2017-07-28:
Currently, `arduino.json` only supports a single path to an Arduino sketch. If a project has multiple sketches, it would be good to be prompted by VSCode to select which one to upload.

In several of my use cases (and I hope others' use cases as well!), a single project can have multiple sketches, for example, in a simple location tracking project, you could have:
- `anchor-firmware/anchor-firmware.ino` for "anchor" hardware
- `tag-firmware/tag-firmware.ino` for "tag" hardware

These sketches would be highly related, sharing the same libraries, Git repository and so on. Hence, depending on what is being worked on, a different sketch needs to be compiled and uploaded.

From my understanding, this is different from #271.

iFreilicht, 2017-08-01:
This is not really possible with the way the Arduino build system works. If you want to have two separate sketches in one project, you'll have to make a separate `arduino.json` file for each of those sketches.

To make this more flexible, this plug-in would require proper support for build tasks, which it doesn't have to my knowledge.

maxime-paquatte, 2017-09-14:
There an other issue opened about that, with a workaround :

https://github.com/Microsoft/vscode-arduino/issues/271

czgtest, 2018-03-06:
@Dismounted  have you got this resolved ?

czgtest, 2018-03-06:
Arduino extension is not support multiple .ino files within single project , Can you have a try @iFreilicht 's suggestion ?

Dismounted, 2018-03-06:
Haven't solved this, but have used a workaround - edit `arduino.json` each time I need to switch to another sketch inside the repo.

czgtest, 2018-03-06:
@ArthurMa1978   ,this issue is duplicate with 481,  Can we close this issue ?

czgtest, 2018-03-06:
#481

riccoyudha, 2019-05-30:
I haven't found a way yet to seamlessly use VS Code to compile my Arduino project with multiple .ino files in a single directory

However, the best way that I have found out to increase my workflow and to avoid gibberish Arduino IDE UI/UX is actually using VSCode as a text editor and use Arduino IDE to upload.

What you need to do :
1. In Arduino IDE, select **File** -> **Preferences** -> Check "**Use External Editor**"
2. In VS Code, select the folder you are working on and just code as usual

Eventually, Arduino IDE will reload the latest update on the .ino files before it uploads to your board.

Hope this helps.

ArthurMa1978, 2019-05-31:
As @iFreilicht  mentioned, our extension leverage the Arduino build system which doesn't support multiple .ino files in one folder / project.
@Dismounted thanks for your suggestion, would put this issue into our backlog since Arduino just released new cli, not sure if it support this feature.

ArthurMa1978, 2019-05-31:
same as #271