microsoft/vscode-arduino #384: Board/Serial Port settings lost between loads

rw950431, 2017-07-30:
I'm liking this new Arduino extension but it seems I have to re-select the board and serial parameters each time I load a folder.  Is there a way for VSCode to remember the board  and comm port settings for each project?   
Issue occurs on Windows 10 + VSCode 1.14.2 + Arduino 1.6.5

yaohaizh, 2017-07-30:
@rw950432, yes, it can. Please refer the readme to see how to persist your settings into the .vscode/arduino.json file.

rw950431, 2017-07-31:
Thanks- that works except I dont see any way to remember the baud rate and other serial parameters- did I miss that too?

czgtest, 2018-03-06:
@rw950431  you can set port and board value in the  .vscode/arduino.json file.

like this:
[image]

czgtest, 2018-03-06:
@rw950431  is it work ?

Sneezry, 2018-04-18:
No more feedback more than one week, close.

closed 2018-04-18 by Sneezry