microsoft/vscode-arduino #386: Disallow main sketch file name to be different from folder by default

iFreilicht, 2017-08-01:
When setting up a new Arduino project, a prompt for the main `.ino` filename appears.

When this filename is not the same as the name of the sketch folder, the Arduino IDE will refuse to open the `.ino` file, stating:

```
The file "my_sketch.ino" needs to be inside a sketch folder named "my_sketch". Create this folder, move the file, and continue?
```

It seems to me there is zero benefit in deviating from the standard procedure there. When initialising a new sketch, it should automatically get the same name as the folder one is in. Renaming it should also rename the project folder. 

This would guarantee that VSCode Arduino projects are always compatible with Arduino IDE projects. It would also alleviate the need for a `"sketch"` field in `arduino.json`.

yaohaizh, 2017-08-02:
This is an issue of old version of Arduino IDE. Please try the latest Arduino IDE version > 1.8.x

iFreilicht, 2017-08-02:
As I said, I'm already using the latest version which is 1.8.3. Or are you talking about a nightly/pre-release version?

liydu, 2018-02-09:
It's an Arduino IDE limitation but will not block the code from compiling in the extension.

closed 2018-02-09 by liydu

ISSac25E, 2022-01-14:
@liydu It does not block code from compiling as long as it does not include any .h or .cpp files from within the same folder. You are still required to have the same folder name as the .ino file for the compiler to find the .h and .cpp files