microsoft/vscode-arduino #389: After laptop wakes up from sleep, Arduino files added to all open VSCodes

JimWatV1, 2017-08-03:
On Win10 w/Creators Update. (15063)

After restarting a sleeping or hibernating laptop with two instances of VSCode running, .vscode/arduino.json and .vscode/c_cpp_properties.json files have been added to all instances.  And the Arduino Examples file is open along with 
(curiously) C:\Users\<username>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\README.md

The folders I have open are not Arduino projects to I have to remove the files each time.

yaohaizh, 2017-08-03:
@JimWatV1  do you have any arduino board connected to the laptop when hibernating?

JimWatV1, 2017-08-03:
Good point.  Sometimes, I do.  I have a Adafruit Feather HUZZAH with ESP8266 WiFi plugged into the docking station, just for power.  I just hibernated while in the doc, and it didn't add the files.  I'll try to keep track when the files get added.

yaohaizh, 2017-08-09:
Fixed in PR#391

closed 2017-08-09 by yaohaizh