microsoft/vscode-arduino #390: Support for "arduino" language id

robertohuertasm, 2017-08-06:
Hi, I'm maintaining [vscode-icons](https://github.com/vscode-icons/vscode-icons) and we want to add support for the `arduino` icon using the language id [vscode-icons#1117](https://github.com/vscode-icons/vscode-icons/issues/1117). 

I've seen that there's another extension using the `arduino` language id (the second one in the store in terms of downloads) and that you decided to drop it in favor of `c++` syntax in #167. I was wondering if you could introduce again the language id for us to support the icon linkage easily. 

To be honest, I'm not very familiar with the effect of adding this `language` attribute so it may not be possible for you to keep both the language id and the syntax separated (language id = arduino && syntax = cpp).

yaohaizh, 2017-08-07:
Current, this extension is leverage the [C++ extension](https://code.visualstudio.com/docs/languages/cpp) for language service, so we have to keep the language id and syntax to C++.  Not sure we can have icon based on the file extension.

robertohuertasm, 2017-08-07:
I thought that maybe you could just provide another id in the `languages` section of your `package.json` as currently, you already have `arduino-output`.  Grammar would still be `cpp` but you would provide `arduino` as a new language id supporting `.ino` and `.pde`.

robertohuertasm, 2017-08-07:
I'll use file extension instead of language id. Don't worry. Thanks!

closed 2017-08-07 by robertohuertasm