microsoft/vscode-arduino #392: includePath not working in c_cpp_properties.json

justin-romano, 2017-08-15:
I'm not sure if this is how its meant to work but it seems wrong.
Really annoying if you trying to build something that uses third party libraries with dependencies.

I have and include folder in my project with the and have added this to the includePath in my c_cpp_properties.json file.
Therefore I expect <test.h> to be included.
but i get...
```
app.ino:1:18: fatal error: test.h: No such file or directory
 #include <test.h>
                  ^
compilation terminated.
```
Now i can do this.
#include "include/test.h" and it works.
But that's no good with you have third party libraries expecting #include <> notation expecting the paths to be working.

I have a folder structure 
```
C:.
│   app.ino
│
├───.vscode
│       arduino.json
│       c_cpp_properties.json
│
└───include
        test.h
```
c_cpp_properties.json

```js
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino",
                "${workspaceRoot}\\include",
                "${workspaceRoot}"
            ],
            "browse": {
                "limitSymbolsToIncludedHeaders": false,
                "path": [
                    "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino",
                    "${workspaceRoot}\\include",
                    "${workspaceRoot}"
                ]
            },
            "intelliSenseMode": "msvc-x64"
        }
    ],
    "version": 2
}
```
test.h
```cpp
class Test
{
  public:
    void printTest()
    {
        Serial.print("test");
    }
};
```
and app.ino

```cpp

#include <test.h>
Test test;

void setup()
{
    Serial.begin(9600);
    test.printTest();
}

void loop()
{
	
}

```

codeworm96, 2017-08-16:
`c_cpp_properties.json` is the config file of the c/c++ extension. It reads the `includePath` config to provide correct intellisense. However, the building done using the Arduino IDE, which follows its own rule to resolve third party libraries.

justin-romano, 2017-08-16:
Well then its kind of useless then for and serious integrated solution.
Basically forget this toy and use Atmel Studio(TM).
Unless of course we should fix it so you can.

liydu, 2018-02-09:
Duplicate #438 .

closed 2018-02-09 by liydu