microsoft/vscode-arduino #393: Automatically configure serial port when only 1 exists

Weetbix, 2017-08-16:
Great work on this extension, works amazingly well!

Would be nice and slightly less tedious if the extension would automatically select the serial port if there is only one port available. Unless you can think of a reason this would be unwanted behaviour?

czgtest, 2018-03-06:
@Weetbix  Thanks for your great suggestion, it's good idea to automatically select the serial port and board type.

czgtest, 2018-03-07:
#439 has the same request.

czgtest, 2018-03-07:
@ArthurMa1978  this issue is duplicate with #439 , #264

czgtest, 2018-03-16:
@Weetbix  Now the Arduino extension already support auto detect new plugged in board , but this feather is only work on windows/mac, not support on Ubuntu.

closed 2018-05-30 by czgtest