microsoft/vscode-arduino #395: Set correct Baud Rate for the board

liydu, 2017-08-22:
Reproduced in v0.24.

For quite some Arduino boards (incl. DevKit), to display serial monitor information correctly, you have to set the Baud Rate to `115200`. By default, the extension is set to `9600`, which is much rare cases. Suggest to set the default Baud Rate to `115200` to reduce an error prone step.

sffamily, 2017-09-25:
increase the priority for this.

closed 2017-09-25 by yaohaizh

pedzed, 2018-01-12:
I am not so happy with this change.

>By default, the extension is set to `9600`, which is much rare cases.

@liydu based on what do you make this conclusion? Every Arduino developer I have seen uses `9600`. The Arduino documentation uses that value as well.

Can I set the baud rate in my `.vscode/arduino.json` file? Every now and then it gets reset to `115200` which is slowly becoming very frustrating.

435vic, 2018-02-20:
You can set the default baud rate in `out/src/serialmonitor/serialMonitor.js`. At the end there are some variables incluiding `SerialMonitor.DEFAULT_BAUD_RATE` the `out` folder is in your arduino extension folder.

bigman73, 2021-09-18:
Why is this issue closed?
Each project can have a different baud rate, not all must use the same global one
Currently the baud rate is reset to 115200 each time VS Code is restarted. The selected value is not persisted.

Can you add a setting to `arduino.json` for `baudRate`?