microsoft/vscode-arduino #407: Upload Using Programmer

jmsaavedra, 2017-09-08:
Apologies if there is an extension setting that I overlooked, but I'm wondering if there is a way to **Upload Using Programmer**. This available in the Arduino app with either `cmd+shift+U` or by holding shift and clicking the Upload button.

There is also a menu selection for which Programmer I am using (_AVRISP MkII_ currently).

Thanks for any info -

icamaster, 2017-09-08:
I am also interested in this. Came here to ask for this and found your post.

icamaster, 2017-09-09:
I think this can be easily fixed by adding here "const args = ["--upload", "--board", boardDescriptor, "--port", dc.port, appPath];" the option "--useprogrammer" as stated here https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc .

RyanBlace, 2017-10-26:
I'd like to see this added too. There will also need to be a way to pick the programmer:

[image]

juankgp, 2017-11-04:
Hi, maybe someone was lucky

guilhermezat, 2018-01-25:
I really need this. ;-)

czgtest, 2018-03-06:
I think this is a nice feature, Arduino extension should be support to select "Programmer" in vscode,  just like selecting Board Type or Port

[image]

czgtest, 2018-05-04:
@icamaster  can you send pull request for your change to support programmer ?

czgtest, 2018-05-10:
vscode-arduino-0.2.15-rc is support to upload using programmer, can you have a try ？
steps:
1.download from here
https://github.com/Microsoft/vscode-arduino/releases/download/v0.2.15-rc/vscode-arduino-0.2.15-rc.vsix
2.open vscode and install vscode-arduino-0.2.15-rc from VSIX
3.Press F1 and select arduino: upload using programmer

czgtest, 2018-05-30:
the latest vscode arduino is support it now, you can upgrade the version 0.2.15 or later.

closed 2018-05-30 by czgtest