microsoft/vscode-arduino #409: Allow end of line to be send using Send Text to Serial Port

mlewand, 2017-09-10:
Currently there's  no way to insert `\n` char to the Serial port, even if copied and pasted from external source VSCode builtin input will strip it.

Sometimes it's a handy to terminate strings using eol.

RyanBlace, 2017-10-26:
Ideally 5 options:
1) Send nothing
2) Send \n (LF)
3) Send \r (CR)
4) Send \r\n (CR & LF)
5) Something even cooler but we can't think of it right now.

[image]

mlewand, 2017-12-20:
Ideally I'd like be able simply to input `\n` or `\r` char(s) into the input that is used to collect about-to-be-sent data.

Your suggestion AFAICS is related to being able to set character that serial input should be followed with, which is slightly different. It will allow you to put new lines, but will require you to put each single line separately.

Whereas having the input element that works with new lines, you can put multiline content before you actually send it.

liydu, 2018-02-09:
Duplicated with #463 as a feature request.

closed 2018-02-09 by liydu