microsoft/vscode-arduino #410: Typical install leads to "Cannot find the OpenOCD from the launch.json debugServerPath property" upon running code

joseortiz3, 2017-09-11:
This is your most typical installation pattern. Windows 10, Default Arduino Desktop IDE installed, default VS Code with no other extensions installed (except those installed as dependencies during vscode-arduino's install).

I'm trying to set up VS Code with this extension. I first installed the default arduino IDE as described in "[Get Started with Arduino](https://www.arduino.cc/en/Guide/Windows)".

I then installed default VS Code, and then installed the vscode-arduino extension default from the extension manager built-in to VS Code.

I'm using an Arduino Uno. It works fine with the Arudino Desktop IDE. When I try to use VSCode "debug", the following occurs with configuration "Arduino":

[image]

No error messages or logs are printed when this occurs.

To me, if OpenOCD were a requirement, it would be listed under [prerequisites](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino).

The configuration "Arduino" has, as default, the following configuration.
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Arduino",
            "type": "arduino",
            "request": "launch",
            "program": "${file}",
            "cwd": "${workspaceRoot}",
            "MIMode": "gdb",
            "targetArchitecture": "arm",
            "miDebuggerPath": "",
            "debugServerPath": "",
            "debugServerArgs": "",
            "customLaunchSetupCommands": [
                {
                    "text": "target remote localhost:3333"
                },
                {
                    "text": "file ${file}"
                },
                {
                    "text": "load"
                },
                {
                    "text": "monitor reset halt"
                },
                {
                    "text": "monitor reset init"
                }
            ],
            "stopAtEntry": true,
            "serverStarted": "Info\\ :\\ [\\w\\d\\.]*:\\ hardware",
            "launchCompleteCommand": "exec-continue",
            "filterStderr": true,
            "args": []
        }
    ]
}

```

yaohaizh, 2017-09-13:
The Arduino UNO is not supported for debugging int VSCode Arduino extension. Please see the readme for  which kind of board suports debugging. Let me know if you have any further questions.

closed 2017-09-13 by yaohaizh

czgtest, 2018-03-07:
please see #446

SunboX, 2018-10-04:
Same for me with Waveshare Driver Board 😢 (Node MCU 1.0 compatible) Works fine with Arduino IDE

afzalmushtaque, 2019-08-03:
If anyone else reading this is trying to "Verify"/"Upload" using VSCode instead of Arduino IDE, please open the command palette by pressing the F1 key and select "Arduino: Verify"/"Arduino: Upload".

leonbrag, 2020-06-03:
I have the same problem with Arduino Uno WiFi rev 2

ChuangRoy, 2021-03-04:
I have same problem with arduino uno .
but I write the debugger path in launch.json ,but  it  isn't work .

This is my launch.json :
`{
    // 使用 IntelliSense 以得知可用的屬性。
    // 暫留以檢視現有屬性的描述。
    // 如需詳細資訊，請瀏覽: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Arduino",
            "type": "arduino",
            "request": "launch",
            "program": "${file}",
            "cwd": "${workspaceFolder}",
            "MIMode": "gdb",
            "targetArchitecture": "arm",
            "miDebuggerPath": "",
            "debugServerPath": "{workspaceFolder}\\OpenOCD\\bin",
            "debugServerArgs": "",
            "customLaunchSetupCommands": [
                {
                    "text": "target remote localhost:3333"
                },
                {
                    "text": "file \"${file}\""
                },
                {
                    "text": "load"
                },
                {
                    "text": "monitor reset halt"
                },
                {
                    "text": "monitor reset init"
                }
            ],
            "stopAtEntry": true,
            "serverStarted": "Info\\ :\\ [\\w\\d\\.]*:\\ hardware",
            "launchCompleteCommand": "exec-continue",
            "filterStderr": true,
            "args": []
        }
    ]
}`

John-WL, 2021-04-08:
can anyone explain to me why this issue was closed? Is this issue solved?