microsoft/vscode-arduino #411: Pre Build command

maxime-paquatte, 2017-09-14:
I need to include text file a string variable (html page to serv). 

A way to do that could be to execute some external command before the build.

Do you thing it's a good choice to add a property in arduino.json & call the given command before the verify task ?

Rlopez, 2017-11-07:
this!
I can't figure out how to to this either, I have a task that does the right thing, but I don't see how to connect it to the verify/upload code (I have that in my .js codebase as preLaunchTask in the launch.json file)

czgtest, 2018-03-07:
@Sneezry  ,  Can you help to investigate it?

czgtest, 2018-05-11:
prebuild will work on 0.2.15 version, so close it

closed 2018-05-11 by czgtest

Rlopez, 2018-05-11:
On thinking about it... Isn't this a bit upside down?
What I mean is that, if instead of vscode-arduino allowing a pre-build
step, wouldn't it be better if VSC could call the vscode-arduino "build" as
part of a build (or pre build) step of itself? That way you get a general
way of doing things and the vscode-arduino codebase doesn't have to care
about the user's particular needs.
So, in VSC, you'd choose "build", which first stop is to generate this
"extra code" and then call vscode-arduino following that (which wasn't
possible at the time of my last comment) . Doesn't that make sense?

Sneezry, 2018-05-14:
We have added a new option to add prebuild command in `arduino.json` in v0.2.15 (see https://github.com/Microsoft/vscode-arduino#options)

@Rlopez do you mean we should just call Arduino IDE build command, and Arduino IDE will handle prebuild action?

Rlopez, 2018-05-14:
@Sneezry I think it is Visual Studio Code that is at fault. Imagine setting up a task (with a sequence of steps) in Visual Studio Code (as can be done today for e.g. pre-debugging) and being able to call a plug-in function (say Arduino build in this case) as one of those steps then the other steps could do whatever extra functionality is needed (for instance calling a bash script)
That way the Arduino code wouldn't need to know about steps outside of itself and the functionality would be much more elastic.