microsoft/vscode-arduino #412: Installing board "packs" in vs code for linux

ivarsu, 2017-09-16:
Whenever i try to install a board pack, vs code opens up the Arduino IDE and outputs "[Starting] Install package - arduino...", nothing gets installed and it keeps opening up Arduino IDEs. When i close the Arduino IDE, vs code outputs "[Done] Installed board package - arduino
". This is not the case...

liydu, 2018-02-09:
Hi @ivarsu , can we know which board you are installing?

ivarsu, 2018-02-09:
hey @liydu ,was trying to install arduino avr boards. This problems seems to no longer be an actual problem. This issue is pretty old as you can see. I kinda ditched vs code for arduino development because i didn't get a response on this issue. But i tried to install the pack now and although i havent tested with an arduino, this install seems to actually have downloaded and installed something. Great work on the extension and editor.

closed 2018-02-09 by ivarsu