microsoft/vscode-arduino #413: Error when trying to debug Arduino M0 Pro

fluentart, 2017-09-20:
Running 1.16 VSCode on Ubuntu Linux, installed the Arduino extension, I can upload code to the board and get data over the serial monitor just fine through VSCode. 
Problem is if I set break points and Press F5, no matter what I do I get an error dropdown at the top of the screen: "**[Error] Cannot read property 'apply' of undefined**"

Any idea what I'm missing?

yaohaizh, 2017-09-20:
@fluentart Could you provide a snapshot for this issue with the console output?

fluentart, 2017-09-21:
When you say snapshot what do you mean? My project folder or a screenshot?
I'll provide as much info as possible:

- VSCode Version: Code 1.16.1 (27492b6bf3acb0775d82d2f87b25a93490673c6d, 2017-09-14T16:26:20.018Z)
- OS Version: Linux x64 4.10.0-33-generic
- Extensions: 

Extension|Author (truncated)|Version
---|---|---
cpptools|ms-|0.12.4
vscode-arduino|vsc|0.2.6

------------------------------------------------------------
OUTPUT:

[Starting] Upload sketch - arduinoM0_ldr.ino
[Warning] Output path is not specified. Unable to reuse previously compiled files. Upload could be slow. See README.
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Sketch uses 16840 bytes (6%) of program storage space. Maximum is 262144 bytes.
Uploading...
Open On-Chip Debugger 0.9.0-g932ec70 (2017-02-15-11:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x000028f4 msp: 0x20002c00
** Programming Started **
auto erase enabled
wrote 32768 bytes from file /tmp/arduino_build_873821/arduinoM0_ldr.ino.bin in 3.583865s (8.929 KiB/s)
** Programming Finished **
** Verify Started **
verified 17104 bytes in 1.409919s (11.847 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
[Done] Uploaded the sketch: arduinoM0_ldr.ino

---------------------------------------------------------------------------------------
.vscode/launch.json contents:

{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Arduino",
            "type": "arduino",
            "request": "launch",
            "program": "${file}",
            "cwd": "${workspaceRoot}",
            "MIMode": "gdb",
            "targetArchitecture": "arm",
            "miDebuggerPath": "",
            "debugServerPath": "",
            "debugServerArgs": "",
            "customLaunchSetupCommands": [
                {
                    "text": "target remote localhost:3333"
                },
                {
                    "text": "file ${file}"
                },
                {
                    "text": "load"
                },
                {
                    "text": "monitor reset halt"
                },
                {
                    "text": "monitor reset init"
                }
            ],
            "stopAtEntry": true,
            "serverStarted": "Info\\ :\\ [\\w\\d\\.]*:\\ hardware",
            "launchCompleteCommand": "exec-continue",
            "filterStderr": true,
            "args": []
        }
    ]
}

--------------------------------------------------------------------------------------------------
.vscode/arduino.json contents:

{
    "board": "arduino:samd:mzero_pro_bl_dbg",
    "port": "/dev/ttyACM0",
    "sketch": "arduinoM0_ldr.ino"
}

-------------------------------------------------------------------------------------------------
.vscode/c_cpp_properties.json contents:

{
    "configurations": [
        {
            "name": "Linux",
            "includePath": [
                "/home/rouan/.arduino15/packages/arduino/hardware/samd/1.6.16/cores/arduino"
            ],
            "browse": {
                "limitSymbolsToIncludedHeaders": false,
                "path": [
                    "/home/rouan/.arduino15/packages/arduino/hardware/samd/1.6.16/cores/arduino",
                    "${workspaceRoot}"
                ]
            },
            "intelliSenseMode": "clang-x64"
        }
    ],
    "version": 3
}

andxu, 2017-09-22:
Please 
1 . set verbose log level : "arduino.logLevel": "verbose"
[image]
2.  delete(if you modified it, please backup it first) arduino.json&launch.json
3.  RE-plug in your device, the arduino.json should be generated for you like:
{
    "board": "arduino:samd:mzero_pro_bl_dbg",
    "port": "COM12",
    "sketch": "app\\app.ino"
}
[image]

4. Press F5, and choose arduino, you should get detailed log from output view and debug console view, give us the logs from both view, thanks.

[image]
[image]

liydu, 2018-02-09:
@fluentart , have you got this resolved?

czgtest, 2018-03-06:
@fluentart  is it fix work now ?

stevenj, 2018-05-29:
I am also getting this error on Linux, trying to debug a SAMD21 board.  Arduino debugging on Linux is broken with this fault.

adumont, 2018-07-23:
Same error here, on:

```
System:    Host: xxxx Kernel: 4.4.0-130-generic x86_64 (ksplice) (64 bit gcc: 5.4.0)
           Desktop: LXDE (Openbox 3.6.1) Distro: Ubuntu 16.04 xenial
```

Board: **AZ3166** (Microsoft MxChip IoT Devkit)

"board": "AZ3166:stm32f4:MXCHIP_AZ3166",

Bus 002 Device 016: ID 0483:374b STMicroelectronics ST-LINK/V2.1 (Nucleo-F103RB)

TuringMachinegun, 2019-03-04:
As I was trying out the new segger-less debugging feature in the Arduino extension, I met the same error, on
```
System:    Host: xxxx Kernel: 4.15.0-45-generic x86_64
           bits: 64
           Desktop: Gnome 3.28.3 Distro: Ubuntu 18.04.2 LTS
```
Board: Adafruit Feather M0

dilin-MS, 2019-03-06:
We currently have no availiable resource to resolve this bug. Move the issue to backlog. Thanks.