microsoft/vscode-arduino #414: Board manager mixes up packages from different sources

emontnemery, 2017-09-24:
With the following settings:
```
"arduino.additionalUrls": "https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json,
                           http://arduino.esp8266.com/stable/package_esp8266com_index.json"
```

Board manager mixes up / merges the ESP8266 packages from esp8266.com and sparkfun.
What is shown in board manager is a merged list of supported boards, and version number from esp8266.com package:
```
SparkFun ESP8266 Boards by SparkFun Electronics
Boards included in this package:
SparkFun Blynk Board - ESP8266, SparkFun ESP8266 Thing, SparkFun ESP8266 Thing - Development Board, 
Generic ESP8266 Module, Olimex MOD-WIFI-ESP8266(-DEV), NodeMCU 0.9 (ESP-12 Module), NodeMCU 1.0 (ESP-12E Module), 
Adafruit HUZZAH ESP8266 (ESP-12), ESPresso Lite 1.0, ESPresso Lite 2.0, Phoenix 1.0, Phoenix 2.0, SparkFun Thing, 
SweetPea ESP-210, WeMos D1, WeMos D1 mini, ESPino (ESP-12 Module), ESPino (WROOM-02 Module), WifInfo, ESPDuino
```

czgtest, 2018-03-06:
This is design by Arduino IDE behavior
Repro steps:
1. only add "Additional Boards Manager URLs" value : thttps://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json
then seleck SparkFun in Boards Manager, you can see 'SparkFun ESP8266 Boards by SparkFun Electronics'
[image]

2. Re-set "Additional Boards Manager URLs" value : 
https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json
http://arduino.esp8266.com/stable/package_esp8266com_index.json

Seleck SparkFun again in Boards Manager,  you can not see 'SparkFun ESP8266 Boards by SparkFun Electronics', only find "esp8266".
[image]

czgtest, 2018-03-07:
@ArthurMa1978 ,This is design by Arduino IDE behavior, can we close it?

emontnemery, 2018-03-07:
@czgtest The issue/bug you have reproduced in Arduino IDE is not same as I reported.
Issue in Arduino IDE
- Only one board package with same name allowed

Issue reported in VS Code plugin
- Two packages with same name are *merged*

czgtest, 2018-03-16:
@Sneezry  can you help to investigate it?

[image]

[image]

Sneezry, 2018-03-27:
@emontnemery If I understand correctly, we should just ignore one board with the same name, but not merge them, right?

emontnemery, 2018-03-27:
You should for sure not merge them.
Only displaying one, just like Arduino IDE is doing, is OK I guess.
A nicer solution might be to display both in the board manager, but only allow installing one at the same time.

Sneezry, 2018-03-27:
@emontnemery OK, thanks for figuring it out.

czgtest, 2018-03-29:
@Sneezry  I can't see any change, the result is the same as before.

[image]

Sneezry, 2018-03-29:
[image]

czgtest, 2018-04-03:
the new version 0.2.12 fix this issue

closed 2018-04-03 by czgtest