microsoft/vscode-arduino #415: Board manager / Library manager do not work with portable Arduino installation

emontnemery, 2017-09-24:
Board manager and library manager do not work when a portable Arduino installation is chosen by arduino.path

(Please let me know if any logs would help to debug this.)

yaohaizh, 2017-10-14:
@eamonoreilly What's your OS version and VSCode version? What's the value of your `arduino.path`?

emontnemery, 2017-10-22:
OS: Windows 10
VSCode: 1.17.2

**Working** `arduino.path` (without `portable` folder):
`"arduino.path": "D:/apps/Arduino/ide/Arduino/arduino-1.8.4"`
Folder `D:/apps/Arduino/ide/Arduino/arduino-1.8.4/portable` does **not** exist, which means
`Arduino15` folder is created under AppData, e.g: `C:\Users\erikm\AppData\Local\Arduino15`

**Non working** `arduino.path` (with `portable` folder):
`"arduino.path": "D:/Apps/Arduino/ide/Arduino_portable_vscode/arduino-1.8.4"`
Folder `D:/apps/Arduino/ide/Arduino/arduino-1.8.4/portable` **does** exist, which means
`Arduino15` folder is **NOT** created under AppData, e.g: `C:\Users\erikm\AppData\Local\Arduino15`
Instead, Arduino settings are stored under `D:/apps/Arduino/ide/Arduino/arduino-1.8.4/portable`
**Note:** To reproduce non working case, rename or delete `Arduino15` folder under AppData/Local if it it exists.

czgtest, 2018-03-05:
It would be better to support portable Arduino installation scenario.
repro steps:
1.) Download the latest Arduino IDE (1.6.4 here) from http://www.arduino.cc/en/pmwiki.php?n=Main/Software
We took the "Windows ZIP file for non admin install" Version.
2.) Unpack the Arduino IDE somewhere
3.) In the Main Arduino IDE Folder, create an Folder named portable

4.) open vscode and install Arduino extension.
5.) set arduino.path in user setting as bellow
[image]
5.) restart vscode and run ardunio:Board manager command

czgtest, 2018-03-29:
After use the new version of Arduino extension. here is the result:
1.if not create portable folder under arduino.path

"arduino.path": "D:/Apps/Arduino/ide/Arduino_portable_vscode/arduino-1.8.5"
[image]

open vscode and run ardunio:Board manager command, then install esp8266 package successfully.
verify the package installation path is in path C:\Users\localname\AppData\Local\Arduino15\packages\esp8266

Press F1 and run ardunio:Library manager command, install Adafruit ESP8266 library  succesfully.
verify the libary installaton path is in C:\Users\localname\AppData\Local\Arduino15\staging\libraries

2.  delete  Arduino15 folder under AppData/Local  and create portable  folder under arduino.path

restart vscode and run ardunio:Board manager command, we also can install esp8266 package successfully.
verify the package installation path is in path D:\Apps\Arduino\ide\Arduino_portable_vscode\arduino-1.8.5\portable\packages\esp8266
and the Arduino15 folder is not created under AppData.

Press F1 and run ardunio:Library manager command, install Adafruit ESP8266 library  succesfully.
verify the libary installaton path is in D:\Apps\Arduino\ide\Arduino_portable_vscode\arduino-1.8.5\portable\staging\libraries

czgtest, 2018-04-03:
the new version 0.2.12 is support portable Arduino installation, Board manager / Library manager /Example can work.

closed 2018-04-03 by czgtest