microsoft/vscode-arduino #419: Nothing is displayed for 'Arduino Board Configuration'

brndmg, 2017-09-30:
When I run the 'Arduino Board Configuration' from the ctrl-shift-p or by clicking the board name in the footer, the tab opens with the title but there isn't any content.

The following error is shown in the Developer console.
```
[Embedded Page] Uncaught Invariant Violation: Element type is invalid: expected a string (for built-in components) or a class/function (for composite components) but got: object. Check the render method of `BoardConfigItemView`. [webview.ts:151]
```

brndmg, 2017-10-02:
I have a working fix for this, I will be submitting a PR shortly.

sid9102, 2017-10-02:
Seeing the same issue on MacOS, please update when this is fixed!

Joseph14078, 2017-10-06:
Can confirm on Win10 with exact same error.

pedzed, 2017-10-07:
Can confirm as well for Debian 9.

```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux testing (buster)
Release:        testing
Codename:       buster
```

godrix, 2017-10-09:
same error occurs in ubuntu 17.04 =(

juankgp, 2017-10-10:
I have the same problem I use Ubuntu 16.04, arduino 1.8.4, code Insigner, I testwith the last version code 1.17, a lot of time ago it works but one day nothing, so please help me

PR #421 (2017-10-10): fixed import, changed class name
I am very new to React. But, by trial and error I discovered that the offending line was: 
```
<Select
```
Along with the message below, which I now assume is synonymous with 'class not found': 
```
Element type is invalid: expected a string 
(for built-in components)
```

And the fact that the class was misnamed as `BoardConfigItemView` threw me for a loop since there is already a class with that name.

Closes #419
closed 2017-10-10 by yaohaizh