microsoft/vscode-arduino #422: Running local unit tests

DonJayamanne, 2017-10-09:
Hi, I'm trying to run run the tests locally and build it (for some PRs), however it doesn't seem to work.
When attempting to run the unit tests I get the following error ```Could not find preLaunch task 'build'.```  

I've removed the ```preLaunchTask``` from the ```launch.json``` file, however when running the tests locally, plenty of them fail. It looks like I need to setup my local environment for some files/directories.

Please could you provide the instructions necessary to test and develop locally. Thanks a lot.

testforstephen, 2017-10-10:
@DonJayamanne The `build` task is actually a gulp task. Did you try to run `gulp build` on cmd manually?

DonJayamanne, 2017-10-10:
Yes thanks, figured that out, looks like the launch.json isn't be used even though its there.

Back to the original question, how do I run the unit tests. They fail at my end, what are the other dependencies required for the unit tests to succeed.

testforstephen, 2017-10-11:
All dependencies are node modules, just need install them by `npm install -g gulp`, `npm install -g typescript`, `npm install`. Besides, need configure `arduino` in the environment variable path for Win/Linux, `arduino.app` in the directory `/Applications` or `/UserHome/Applications/` for Mac.

There are two ways to run unit tests.

- Run unit test by selecting Launch Tests from launch.json. The launch.json is valid, it works in my PC.
- Run unit test by cli command. You should run `gulp build` to compile ts scripts first, then run `npm test --silent`. The travis ci uses this way to run unit test. (refer to .travis.yml file)

DonJayamanne, 2017-10-11:
Installing the node dependencies was obvious. As mentioned earlier, I don't think the launch.json is being used. Will submit a PR to fix that, else it should be removed as it doesn't work at all.

>Besides, need configure arduino in the environment variable path for Win/Linux, arduino.app in the directory /Applications or /UserHome/Applications/ for Mac.

This is what I was after. Thanks.

closed 2017-10-11 by DonJayamanne