microsoft/vscode-arduino #427: "Cannot find the sketch file" error when verifying/uploading code

EnforcerZhukov, 2017-10-17:
I created an .ino file in VSCode using this Arduino extension. Everything seems to work fine, linked the extension to the folder where I have Arduino IDE 1.8.5 extracted. Intellisense is working fine. I can switch boards and open serial monitor.

The problem comes when I try to verify or upload the sketch to the board, then a popup message tells:

`Cannot find the sketch file.`

Here's a screenshot from the VSCode window when the error appears:
[image]

* OS: Ubuntu 16.04 xenial, x86_64 Linux 4.10.0-37-generic
* VSCode 1.17.1
* Arduino IDE 1.8.5 for Linux

yaohaizh, 2017-10-18:
@EnforcerZhukov Wthat is the content in the `arduino.json` file?

EnforcerZhukov, 2017-10-18:
Where is that file?

starock, 2017-11-22:
Where is the fukking arduino.json?!   (MAC OS 10.12)

liydu, 2018-02-09:
@starock , the `arduino.json` file is located in the exact project folder within `.vscode`. You can use `ls -al` to see it.

czgtest, 2018-03-05:
config the sketch value in arduino.json, for example : "sketch": "GetStarted.ino",

[image]

czgtest, 2018-03-07:
@EnforcerZhukov  have you got this issue solved?

EnforcerZhukov, 2018-03-07:
Sorry, I'm not currently working with Arduino. I'll try when I get back to it.
Thanks!

EnforcerZhukov, 2018-03-07:
Hi again, thanks for the help.

Tried with a random sketch specifying the "sketch" in the arduino.json file and seems to work fine now!

[image]

Sneezry, 2018-03-07:
@EnforcerZhukov great! May I close this issue if it has been resolved?

webdevbrian, 2019-06-11:
For people still getting this issue: I needed to create a workspace for the .ino file I had. VSCode needs to create a hidden directory with the aforementioned file in it.  Not doing so results in the above error. I created a vscode along the .ino file I had (in a separate folder) and saved the workspace. Setup everything in VSCode and everything worked as expected.

Just trying to help people coming from google like I did.