microsoft/vscode-arduino #429: Verification always gives an [Error] Exit with code=undefined

my-poi, 2017-10-20:
Hello!

Verification gives an error:
```
[Starting] Verify sketch - projects/001.ino
[Error] Exit with code=undefined
```
The code is a blink template.

Arduino.json
```
{
    "port": "COM6",
    "board": "arduino:samd:mzero_pro_bl_dbg",
    "sketch": "projects\\001.ino",
    "output": "build"
}
```
c_cpp_properties.json
```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:\\Users\\maciej.tokarz\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.6.16\\cores\\arduino"
            ],
            "browse": {
                "limitSymbolsToIncludedHeaders": false,
                "path": [
                    "C:\\Users\\maciej.tokarz\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.6.16\\cores\\arduino",
                    "${workspaceRoot}"
                ]
            },
            "intelliSenseMode": "msvc-x64"
        }
    ],
    "version": 3
```
Settings
```
    "arduino.path": "C:/Program Files (x86)/Arduino",
    "arduino.logLevel": "verbose"
```
The most interesting thing is that despite the Verify error, Upload works.

Regards,
Maciej

my-poi, 2017-10-21:
The solution was to reset Windows to factory settings.
M.

closed 2017-10-21 by my-poi

yoppy-tjhin, 2017-12-29:
Hi,

I encounter the same problem, and could not resolve it.
I am using Win 8. How did you actually reset Windows to factory settings?
Thank you for sharing.

RJPalombo, 2018-03-14:
First time trying this out and I received the same error (on Mac). For me the reason was I copied the settings from the instructions. I commented out all but  _"arduino.enableUSBDetection": true_ and everything is working now. I now see that you are only supposed to use these settings if you want to override the defaults - which for me the defaults work fine.

FriedCircuits, 2018-03-21:
I get this error on Win10 for both verify and upload.

arkadiusz-wieczorek, 2018-03-29:
@RJPalombo Exactly the same in the case of Ubuntu 16.04, I've added in User Settings: 
`"arduino.enableUSBDetection": false`

alkhimey, 2018-04-05:
Just installed Arduino extension and can confirm that I have the same problem.

jraviotta, 2018-04-13:
On OSX, I get the same error.

driedell, 2018-04-19:
I was trying on Windows 10 and was getting the same error when using the Arduino IDE downloaded from the Windows 10 app store. But when I downloaded the Arduino IDE directly from https://www.arduino.cc/ and uninstalled the Win10 app store version, it worked.

mtzfactory, 2018-05-05:
Hi,

I had the same problem in my **Mac**, but now it's working properly. I changed a bit the _settings.json_ file:

```json
{
    "arduino.path": "/Applications",
    "arduino.commandPath": "Contents/MacOS/Arduino",
    "arduino.additionalUrls": "",
    "arduino.logLevel": "info"
}
```

jraviotta, 2018-05-05:
Seems to be a documentation issue on the ardiuno path. Also, documentation should reflect that the path can be set in settings.jaon and the project vscode.