microsoft/vscode-arduino #434: [Board Manager] Arduino Extension doesn't load valid boards

legionaryu, 2017-10-25:
I've downloaded the [breadboard-1-6-x.zip](https://www.arduino.cc/en/uploads/Tutorial/breadboard-1-6-x.zip) from the URL [https://www.arduino.cc/en/Tutorial/ArduinoToBreadboard](https://www.arduino.cc/en/Tutorial/ArduinoToBreadboard) and it loads correctly in the Arduino IDE but is ignored by the VS Code Arduino Extension.

Debugging the Arduino Extension from the source I've found out that it expects a "platform.txt" file in the same folder as the "boards.txt".

Is this the desired behavior? Shouldn't the Arduino Extension behave like the Arduino IDE itself?

Thanks in advance!

JulianGmp, 2017-11-17:
I'm having a similar issue, I downloaded the package for the Espressif ESP 8266 and installed it,  according to their installation guide in my documents directory (`D:\Users\Julian\Documents\Arduino\hardware\esp8266com\esp8266`).
However when trying to verify/upload I get an error saying the package is unknown and it exists with `code=3`

Output:
```
[Starting] Upload sketch - example.ino
[Warning] Output path is not specified. Unable to reuse previously compiled files. Upload could be slow. See README.
Looking for library 'user32'
Adding paths from jna.library.path: null
Trying user32.dll
Found library 'user32' at user32.dll
DPI detection failed, fallback to 96 dpi
Looking for library 'shell32'
Adding paths from jna.library.path: null
Trying shell32.dll
Found library 'shell32' at shell32.dll
Looking for library 'Ole32'
Adding paths from jna.library.path: null
Trying Ole32.dll
Found library 'Ole32' at Ole32.dll
Konfiguration wird geladen... (Loading Config)
Pakete werden initialisiert... (Initialising Packages)
Boards werden vorbereitet... (Preparing Boards)
Fehler: esp8266: Unbekanntes Paket (Error: esp8266: unknown package)
[Error] Exit with code=3
```
_Note: I edited some translations in, since some output is in German (my system language)._

I tried manually adding the path(s) to the package in my config, trying the multiple paths, but without any luck
```json
"arduino.additionalUrls": [
    "D:/Users/Julian/Documents/Arduino/hardware/",
    "D:/Users/Julian/AppData/Local/arduino15/packages",
    "D:/Users/Julian/AppData/Local/arduino15/staging/packages",
    "D:/Users/Julian/Documents/Arduino/hardware/esp8266com/esp8266"
],
```

For now I just have intellisense helping me out and I verify and build using the arduino "IDE".

czgtest, 2018-03-07:
@Sneezry  Can you help to investigate ?

sslupsky, 2018-09-22:
Attempted to load Arduino SAMD boards package and received "selected board is not available" error.  Successfully loaded the package using the Arduino IDE.

Reynolds-Beta, 2020-05-24:
> I've downloaded the [breadboard-1-6-x.zip](https://www.arduino.cc/en/uploads/Tutorial/breadboard-1-6-x.zip) from the URL https://www.arduino.cc/en/Tutorial/ArduinoToBreadboard and it loads correctly [...]

Having the exact same problem. Arduino IDE (1.8.12) shows my custom board definition for **ATMega328P on a Breadboard (8Mhz internal clock)**, but VSCode doesn't list this board.

As per the Arduino IDE, I have an additional boards.txt in:
C:\<user>\Documents\Arduino\hardware\breadboard\avr

```
atmega328bb.name=ATmega328 on a breadboard (8 MHz internal clock)

atmega328bb.upload.protocol=arduino
atmega328bb.upload.maximum_size=30720
atmega328bb.upload.speed=57600

atmega328bb.bootloader.low_fuses=0xE2
atmega328bb.bootloader.high_fuses=0xDA
atmega328bb.bootloader.extended_fuses=0x05

atmega328bb.bootloader.file=atmega/ATmegaBOOT_168_atmega328_pro_8MHz.hex
atmega328bb.bootloader.unlock_bits=0x3F
atmega328bb.bootloader.lock_bits=0x0F

atmega328bb.build.mcu=atmega328p
atmega328bb.build.f_cpu=8000000L
atmega328bb.build.core=arduino:arduino
atmega328bb.build.variant=arduino:standard
atmega328bb.bootloader.tool=arduino:avrdude
atmega328bb.upload.tool=arduino:avrdude
atmega328bb.build.board=AVR_ATMEGA328BB
```

agucova, 2021-01-30:
I'm trying to load a Sparkfun Pro RF board using `arduino.additionalUrls` (which actually loads some boards) but it doesn't load the Pro RF to the list integrated in VSCode. (It works fine on Arduino IDE).

dasfdlinux, 2021-02-03:
> Having the exact same problem. Arduino IDE (1.8.12) shows my custom board definition for ATMega328P on a Breadboard (8Mhz internal clock), but VSCode doesn't list this board.
>
>As per the Arduino IDE, I have an additional boards.txt in:
C:<user>\Documents\Arduino\hardware\breadboard\avr

Same here. My board shows up in Arduino IDE (1.8.13) but not in VSCode. I am not using a custom platform.txt file as my board is based on a current Arduino variant.

<pre><code>mega2560_8MHz.name=Arduino Mega 2560 (8MHz External Crystal)

mega2560_8MHz.upload.tool=arduino:avrdude
mega2560_8MHz.upload.protocol=wiring
mega2560_8MHz.upload.maximum_size=253952
mega2560_8MHz.upload.maximum_data_size=8192
mega2560_8MHz.upload.speed=115200

mega2560_8MHz.bootloader.tool=arduino:avrdude
mega2560_8MHz.bootloader.low_fuses=0xFF
mega2560_8MHz.bootloader.high_fuses=0xD8
mega2560_8MHz.bootloader.extended_fuses=0xFD
mega2560_8MHz.bootloader.file=stk500v2/stk500boot_v2_mega2560.hex
mega2560_8MHz.bootloader.unlock_bits=0x3F
mega2560_8MHz.bootloader.lock_bits=0x0F

mega2560_8MHz.build.mcu=atmega2560
mega2560_8MHz.build.f_cpu=8000000L
mega2560_8MHz.build.core=arduino:arduino
mega2560_8MHz.build.variant=arduino:mega
mega2560_8MHz.build.board=AVR_MEGA2560V</code></pre>

If I tried to burn a bootloader my boards.txt would fail since it's looking for the bootloader.file in my custom directory, but I just use the burn bootloader command for the purpose of setting fuses.

jaajuu, 2024-04-07:
Hi,
I came across this ancient thread while running in to the same/similar issue.
current setup:
VSCode Version: 1.87.2
Arduino: 1.8.19
Trying to compile a ESP8266 project in a new VM with VSCode, that was working fine in the old development environment. I have installed the esp8266 via Arduino IDE and all the needed libraries. And eventually only error left was: "Error: esp8266: unknown package"

However it turned out, that once I compiled the project with ArduinoIDE also the VSCode version worked afterwards. 
Go figure...