microsoft/vscode-arduino #437: Identifier is undefined

dafzor, 2017-10-27:
All Arduino functions in Visual Studio code show underlined in red with the following error:
```file: 'file:///c%3A/Users/daf/Documents/Arduino/ex2_ledseq/ex2_ledseq.ino'
severity: 'Error'
message: 'identifier "pinMode" is undefined'
at: '10,5'
source: ''
```

Adding `#include <Arduino.h>` fixes the error but shows a new one:

```file: 'file:///c%3A/Users/daf/Documents/Arduino/ex2_ledseq/ex2_ledseq.ino'
severity: 'Info'
message: 'cannot open source file "stdlib.h" (dependency of "Arduino.h")'
at: '1,1'
source: ''
```
This error shows on two computers I tested, around one to two weeks ago it did not occur. Verification and Upload still work without regardless of the error above.

redwarff, 2017-10-29:
I am having the exact same problem

Kulbhushan-Chand, 2017-11-08:
Same problem here.
Is there a way not to write the basic include files in the beginning, just like we do in arduino official IDE?

JebinMatthew25, 2017-11-20:
Yep exactly, but the code can be uploaded without problems, still it's annoying

yaohaizh, 2017-11-21:
Duplicate with https://github.com/Microsoft/vscode-arduino/issues/438

Please see the related issue.

closed 2017-11-21 by yaohaizh