microsoft/vscode-arduino #438: How do I set up intellisense so it picks all the libraries?

mundodisco8, 2017-11-04:
To my knowledge, the Arduino IDE checks for libraries in 4 places: the main Arduino installation path (in windows c:\program files.\...), any board-specific version of the libraries for this board ("...\appdata\local\..."), your sketchbook location and a "libraries" folder for that particular sketch. Ideally, you would install your libraries in your sketchbook location, so they are available for all your projects.

I've just installed VS Code and the Arduino plugin, and it seems to be able to know that my sketchbook location is not the default, as I can see the examples of the libraries that I've installed there, and I can compile my sketch. In any case, I get tens of intellisense errors, as intellisense seems incapable to follow the paths for the includes. I can add the sketchbook location libraries to the c_cpp_properties.json, and either add all the libraries used there (painfull) or add the base "libraries" folder and then tweak the includes to have partial paths (less painfull), but even in these cases, intellisense will complain that it can't find any of the libraries this files might use. And I would have to do it for all my projects, one by one.

What am I doing wrong? why doesn't this plugin pick the usual library locations? Is it normal to have 100s of intellisense errors? is that really the way it's meant to be?

Sorry, I'm new to Visual Studio and Visual Studio Code, and I'm quite sure that the problem it's me not being able to find out the logic behind them, but I can't find any info on the internet either.

I'm using windows 10 64 bits, and my arduino Ide is at the latest version. The relevant part of my  c_cpp_settings.json of this project is:

```{
            "name": "Win32",
            "includePath": [
                "C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.11.25503/include/*",
                "C:/Program Files (x86)/Windows Kits/10/Include/10.0.16299.0/um",
                "C:/Program Files (x86)/Windows Kits/10/Include/10.0.16299.0/ucrt",
                "C:/Program Files (x86)/Windows Kits/10/Include/10.0.16299.0/shared",
                "C:/Program Files (x86)/Windows Kits/10/Include/10.0.16299.0/winrt",
                "${workspaceRoot}",
                "C:\\Users\\Joel\\AppData\\Local\\Arduino15\\packages\\SODAQ\\hardware\\samd\\1.6.14\\cores\\arduino"
            ],
            "defines": [
                "_DEBUG",
                "UNICODE"
            ],
            "intelliSenseMode": "msvc-x64",
            "browse": {
                "path": [
                    "C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.11.25503/include/*",
                    "C:/Program Files (x86)/Windows Kits/10/Include/10.0.16299.0/um",
                    "C:/Program Files (x86)/Windows Kits/10/Include/10.0.16299.0/ucrt",
                    "C:/Program Files (x86)/Windows Kits/10/Include/10.0.16299.0/shared",
                    "C:/Program Files (x86)/Windows Kits/10/Include/10.0.16299.0/winrt",
                    "${workspaceRoot}",
                    "C:\\Users\\Joel\\AppData\\Local\\Arduino15\\packages\\SODAQ\\hardware\\samd\\1.6.14\\cores\\arduino"
                ],
                "limitSymbolsToIncludedHeaders": false,
                "databaseFilename": ""
            }
        }```

![2017-11-04_1307](https://user-images.githubusercontent.com/8402707/32405654-8059e4bc-c161-11e7-8d94-691459148d2b.png)

yaohaizh, 2017-11-04:
Please see here to see whether help resolve this issue?  
https://github.com/Microsoft/vscode-arduino/wiki/FAQ

By default, the Arduino extension doesn't pick up all the head files. You can either pop up the command "Arduino: Library Manager" to include the header file to your paths.

mundodisco8, 2017-11-06:
I try to avoid the arduino Library manager as much as I can. In any case, I just deleted VSC completely, including the extensions and settings folder, and stated again. I installed MS c/c++ extension first, and then the Arduino one. I opened an example, the Blink sketch, and it was full of undefined identifiers (pinMode, digitalWrite, HIGH,... anything arduino related). I selected a board (arduino uno), to create the arduino.json file and the c_cpp_properties.json, but the undefines still remain. This should not be undefined by default, the path should include whatever is required so a normal sketch doesn't present any errors (it compiles, but still shows "problems", that doesn't make sense and makes debugging very complicated).

I checked what you said about including the libraries from the manager. the blink sketch doesn't require to add any other libraries, but I added one, SD. The correct path was added to c_cpp...json, which is nice, but it complains of being unable to find the dependencies of the SD library. The arduino IDE has a set of folders to look for libraries, the plugin should now this and search there. 

What are the folders that I am supposed to add to the path so I can compile standard Arduino sketches, without any added libraries?

mundodisco8, 2017-11-06:
Hi,

I just open an example, the blink example, and used the arduino Uno as board. That way, my c_cpp_properties.json has just these two folders:
```"C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino",
"${workspaceRoot}"```

With that, all the Arduino-defined functions and variables are highlighted as undefined (pinMode, digitalWrite, HIGH,...).

It may be that I don't get how this thing is supposed to work, but I expect from the Arduino plugin to get all these methods by default. I also don't understand why doesn't it search for libraries where the Arduino IDE is expecting to find libraries (that is, the installation folder, the roaming/local/... folder where your boards are installed and the sketchbook location). These folders should be added by default. In any case, even if I add them, I can't get it to work. Yes, it compiles, but it's impossible to use this in a medium sized sketch without getting dozens of errors that make debugging a nightmare. 

If my understanding is correct, only the folders in the includePath parameter are used for intellisense, but the search in them is not recursive. I don't know what folders do I have to add to get all the different definitions and methods and whatnot from Arduino working, but I suspect that it could be again dozens of them, not including all the folders I have to include to make a library work if it has dependencies... Is there a way to get Intellisense working on the Arduino plugin or am I doing something that is simply not achievable?

bandrews, 2017-11-07:
I've been fighting the same thing;  since the intellisense parser changed to requiring exact paths from being willing to follow recursive directories, it's basically impossible to set up all the include paths necessary for Arduino usage.

I've been able to add all the libraries to the browse path, which inexplicably _does_ work with recursion.  Then, VS Code will suggest folders that are missing from the include path in the light bulb bubble.  But I have to play whack-a-mole going deep into the bowels of included microcontroller library structures to get everything added.  And the suggestions stop working after you pick only one, so you end up with lots of missing libraries and no way to discover the necessary paths for them...

This really needs to be fixed by the C/C++ extension authors, but judging from their issues database they've seemed stubbornly resistant to feedback requesting the return of recursive search paths.  :(

mundodisco8, 2017-11-08:
That's a shame, let's see how things develop, but it looks like a dealbreaker, at least for me at the moment.  :(

GamMou, 2017-11-11:
Also been fighting this issue. I have been trying to use the LiquidCrystal library for a project of mine, First it's annoying and unwieldy trying to point to that file - let alone dig around to then find (and point to) the dependencies within that. The c_cpp_properties.json file is becoming a mess trying to keep - I'm rather discouraged to work on a bigger project which might need more libraries. 

Any other solutions or alternatives for now? It's a shame because I love Visual Studio Code and it would be perfect to add another type of project to it (I use it for Unity (C#), Java, and Web Development with zero issues)

yaohaizh, 2017-11-13:
For please this value to your vscode setting file to see whether fix the intellisense issue: 

```json
"C_Cpp.intelliSenseEngine": "Tag Parser",
``

GamMou, 2017-11-14:
This partially works (In that it stops the problems showing up in my test.ino file; and it seems I can show IntelliSense for methods not affected by problematic includes deeper down the rabbit hole) 

As you can see 

[image]

It will suffice for now - At least I can program for (and build) to my Arduino from the comfort of VS Code :) Looking forward to see even tighter / full integration soon.

Thanks @yaohaizh

JulianGmp, 2017-11-17:
I was in a similar situation trying to set up vscode for programming ESP chips (using Arduino).
I got intellisense to work, but I haven't done too much testing so this solution might not be final one.

But it does give me preview for parameters in the FastLED Library, which is installed under `D:\Users\Julian\Documents\Arduino\libraries\FastLED`.
[image]

**Stuff I did:**
 - _Force Intellisense to use the "Tag Parser" in the User Settings:_
```json  
"C_Cpp.intelliSenseEngineFallback": "Disabled",
"C_Cpp.intelliSenseEngine": "Tag Parser",
```

 - *Set up the c_cpp_properties.json*
```json
{
    "name": "Arduino",
    "includePath": [
        "${workspaceRoot}"
    ],
    "browse": {
        "path": [
            "D:/Program Files (x86)/Arduino/hardware",
            "D:/Program Files (x86)/Arduino/libraries",
            "D:/Users/Julian/Documents/Arduino/libraries/",
            "D:/Users/Julian/Documents/Arduino/hardware/",
            "${workspaceRoot}"
        ],
        "limitSymbolsToIncludedHeaders": true,
        "databaseFilename": ""
    },
    "intelliSenseMode": "msvc-x64"
}
```
that way, only the Tag Parses will be used as the engine and the config in the c_cpp_properties.json becomes a lot simples, since intellisense reads directories recursively now. That way I also got the engine to read libraries I installed manually, since they are placed in my homedirectory (`D:/Users/Julian/Documents/Arduino/libraries/`). Same goes for boards I installed, like the ESP8266, which has its pakage sit in `"D:/Users/Julian/Documents/Arduino/hardware/"`.

I'm gonna try to config the Arduino extension from Microsoft as well now, if I run into some issues I can fix I will comment them here.

dafzor, 2017-11-21:
I have not tried the tag parser setting suggested here, but had come up with a solution which was to simply add all the include paths by hand.

For a basic hello world program on a UNO like this:
```
#include <Arduino.h>
#define LED_PIN 10

void setup() {
    pinMode(LED_PIN, OUTPUT);
}

void loop() {
    digitalWrite(LED_PIN, HIGH);
    delay(1000);
    digitalWrite(LED_PIN, LOW);
    delay(1000);
}
```
I need a C_Cpp_properties.json like this:
```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "${workspaceRoot}",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino",
                "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr\\avr\\include",
                "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr\\lib\\gcc\\avr\\4.9.2\\include",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega"
            ],
            "browse": {
                "limitSymbolsToIncludedHeaders": false,
                "path": [
                    "${workspaceRoot}",
                    "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino",
                    "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr\\avr\\include",
                    "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr\\lib\\gcc\\avr\\4.9.2\\include",
                    "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega"
                ]
            },
            "intelliSenseMode": "msvc-x64"
        }
    ],
    "version": 3
}
```
Maybe changing it so that Arduino: Board Config actually includes all the needed paths could be a start for trying to fix the issue.

InRiPa, 2017-11-22:
@dafzor Thanks. I adjusted it for linux and it works. At least I got autocompletion and intellisense working. In case someone else needs it, here is my config:

    {
        "configurations": [
            {
                "name": "Linux",
                "includePath": [
                    "/usr/lib/gcc/avr/7.2.0/include",
                    "/home/<user>/.arduino15/packages/arduino/hardware/avr/1.6.20/cores/arduino",
                    "/usr/avr/include/",
                    "/home/<user>/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/avr/include",
                    "/usr/share/arduino/hardware/archlinux-arduino/avr/variants/mega"
                ],
                "browse": {
                    "limitSymbolsToIncludedHeaders": false,
                    "path": [
                        "/usr/lib/gcc/avr/7.2.0/include",
                        "/home/<user>/.arduino15/packages/arduino/hardware/avr/1.6.20/cores/arduino",
                        "/usr/avr/include/",
                        "/home/<user>/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/avr/include",
                        "/usr/share/arduino/hardware/archlinux-arduino/avr/variants/mega"
                    ]
                },
                "intelliSenseMode": "clang-x64"
            }
        ],
        "version": 3
    }

heyAyushh, 2017-11-28:
Somebody Please tell this for mac too

carlhung, 2017-12-16:
I can't find the path for Mac

stormbytes, 2017-12-17:
Same issue.

Added '#include <Arduino.h>' to the top of my sketch which makes the errors go away. Not sure about intellisense or anything else. Anyone at Platformio working on this?

JulianGmp, 2017-12-17:
@stormbytes I think that is caused because the Arduino.h file isn't included by default in Visual Studio's mind. The Arduino Software obviously includes their core when compiling. 
Adding the `#include <Arduino.h>` or the `#include <Esp.h>` when using ESP Controllers won't cause issues though. It's a bit annoying but oh well, better than having to use the Arduino "IDE"

stormbytes, 2017-12-17:
JulianGmp what you're saying makes sense though Platformio team should have considered this when developing the VS extension. Having to add the header to each sketch is completely redundant and means that everyone who uses this extension is bound to run into this exact problem. It will do for now. I hope they fix this is the next update.

JulianGmp, 2017-12-17:
@stormbytes I'm not using PlatformIO, I'm using the [Arduino Extension from Microsoft](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino)

yaohaizh, 2017-12-18:
The official Arduino IDE will include the <Arduino.h> behind in their tooling. But for VSCode Arduino extension, we have to include explicitly, just as normal C/C++ file to make the intellisense work. But you doesn't need this header file for build/upload.

dafzor, 2017-12-18:
Just leaving the info here that even if you setup the include paths the intellisense will still not work 100%. For example, if you include SD.h library and a line like this `File myfile = SD.open("myfile");` it will show several type definition errors even though the sketch will compile and run correctly.

So I ended up moving to http://www.visualmicro.com until the vscode extension gets fixed.

DavidKarlas, 2018-01-06:
On macOS, I added `#include <Arduino.h>` to beginning of `.ino` file and I filled `c_cpp_properties.json` content with:
```
{
    "configurations": [
        {
            "name": "Mac",
            "includePath": [
                "${workspaceRoot}",
                "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino",
                "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/mega",
                "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include",
                "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/lib/gcc/avr/4.9.2/include"
            ],
            "browse": {
                "limitSymbolsToIncludedHeaders": false,
                "path": [
                    "${workspaceRoot}",
                    "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino",
                    "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/mega",
                    "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include",
                    "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/lib/gcc/avr/4.9.2/include"
                ]
            },
            "intelliSenseMode": "clang-x64"
        }
    ],
    "version": 3
}
```
Notice that `#include <Arduino.h>` line shows error, but you can ignore it and have Intelisense working for rest of file fine...

heroboy, 2018-01-10:
I think you can run `arduino_debug.exe`, compile a sketch. You can look the console window for it how to call `avr-g++` to know the predefined macros and include path. Finally my json file look like this:
```json
{
    "configurations": [
        {
            "name": "Win32",
            "defines": [
                "F_CPU=16000000L",
                "ARDUINO=10805",
                "ARDUINO_AVR_UNO",
                "ARDUINO_ARCH_AVR"
            ],
            "includePath": [
                "D:\\tools\\arduino-1.8.5\\hardware\\arduino\\avr\\cores\\arduino",
                "D:\\tools\\arduino-1.8.5\\hardware\\arduino\\avr\\variants\\standard",
                "D:\\tools\\arduino-1.8.5\\hardware\\tools\\avr\\avr\\include",
                "D:\\tools\\arduino-1.8.5\\hardware\\tools\\avr\\lib\\gcc\\avr\\4.9.2\\include"
            ],
            "browse": {
                "limitSymbolsToIncludedHeaders": false,
                "path": [
                    "D:\\tools\\arduino-1.8.5\\hardware\\arduino\\avr\\cores\\arduino",
                    "D:\\tools\\arduino-1.8.5\\hardware\\arduino\\avr\\variants\\standard",
                    "D:\\tools\\arduino-1.8.5\\hardware\\tools\\avr\\avr\\include",
                    "D:\\tools\\arduino-1.8.5\\hardware\\tools\\avr\\lib\\gcc\\avr\\4.9.2\\include"
                ]
            },
            "intelliSenseMode": "msvc-x64"
        }
    ],
    "version": 3
}
```

raffaeler, 2018-01-13:
I also retried today to configure the support for intellisense and it doesn't work.
After configuring the includePath, now it request WProgram.h even if the Arduino installation is totally updated.
I don't want to waste time and give up.

clouds56, 2018-01-20:
If you meet something like `identifier "Serial" not defined`, also add "USBCON" into `defines` in `configurations`
See [cpptools document](https://github.com/Microsoft/vscode-cpptools/blob/master/Documentation/Getting%20started.md) for more details.

I'm not sure if there're other flags like this, while I hope the extension could do this automatically for us, since we have done setup the "Arduino Board Configuration".

raffaeler, 2018-01-20:
Nope, in my case the problem is the editor. No way to configure properly the intellisense and get rid of the squiggles on the includes

clouds56, 2018-01-20:
Which header is trying to request "WProgram.h"? In my linux package, there's no WProgram.h and no header's including it.

dafzor, 2018-01-20:
quick search trough the headers shows this bit of code

`* The WProgram.h file, which provides declarations for the Arduino API,
  has been renamed to Arduino.h. To create a library that will work in
  both Arduino 0022 and Arduino 1.0, you can use an #ifdef that checks
  for the ARDUINO constant, which was 22 and is now 100.  For example:

  #if defined(ARDUINO) && ARDUINO >= 100
  #include "Arduino.h"
  #else
  #include "WProgram.h"
  #endif`

So seems you're using an old library that doesn't properly account for
newer versions of the API.

raffaeler, 2018-01-20:
The library is DHT.h from Adafruit and is updated.
The DHT.h file specifies the same preprocessor definitgions you wrote, but it complains
[image]

If I right click the DHT.h and navigate to definition, the DHT.h file gets opened.
Arduino IDE version 1.8.2
DHT unified 1.0.0
DHT sensor library 1.3.0

clouds56, 2018-01-20:
It's also about defines, the ARDUINO macro is not defined, add 
```
defines: [ "ARDUINO=100" ],
```
to your [c_cpp_properties.json](https://github.com/Microsoft/vscode-cpptools/blob/master/Documentation/LanguageServer/c_cpp_properties.json.md)

I'm assuming you're using this [DHT.h](https://github.com/adafruit/DHT-sensor-library/blob/master/DHT.h), and your ARDUINO version is 100, change it if necessary.

raffaeler, 2018-01-20:
Looks far better now, thanks!
I am adding other include paths and then will try connecting the devices.

I don't understand why the vscode extension does not provide somthing to get this easier. Also the doc does is not clear about the settings needed to make it work.

mundodisco8, 2018-02-02:
Still, all the solutions force you to use the tag parser, as the default mode doesn't work. I don't think the issue is solved, as the default mode doesn't work. Even including all the individual paths, I still get errors of undefined stuff that is clearly defined in libraries I'm including.
Having to go back a mode that has less features in order for it to work doesn't seem like a solution to me, but more like a patch while we work on the issue (something that I'm super grateful to have the option to use as a fallback, don't get me wrong). 
That's why I think the issue should be open. I know that intellisense is not part of this plugin, but you should be either able to talk to the c/c++ plugin team in order to make them aware of this type of issues, or change the c_cpp_properties.json on the fly, based on the selected board, to add the barebones "browse.path".

mundodisco8, 2018-02-02:
Another example:
I get errors on all the register names and bit names used to manipulate them. They are defined in "avr/io.h" (location dependant on your architecture), which is basically a bunch of #ifdefs. I understand that is not being picked because the relevant word for my board is not being defined (whereas the Arduino IDE defines all the relevant words), much like in the case of raffaeler with the Wprogram.h library. How can we overcome this? is there a list of stuff that is known to be defined by the IDE that is not being properly passed to intellisense?

aster94, 2018-02-03:
+1 this should be the default so the end-user doesn't have to worry about these path
hope to see in the next release!

jlj-ee, 2018-02-12:
Cross-reference: Microsoft/vscode-cpptools#743
For the Arduino Uno, I have found that the following settings work well:
```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "${workspaceRoot}",
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino",
                "C:/Program Files (x86)/Arduino/hardware/tools/avr/avr/include",
                "C:/Program Files (x86)/Arduino/hardware/tools/avr/lib/gcc/avr/4.9.2/include",
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/variants/standard",
            ],
            "defines": [
                "_DEBUG",
                "UNICODE",
                "__AVR_ATmega328P__",
                "__AVR_ENHANCED__"
            ],
            "intelliSenseMode": "clang-x64",
            "browse": {
                "path": [
                    "${workspaceRoot}",
                    "C:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino",
                    "C:/Program Files (x86)/Arduino/hardware/tools/avr/avr/include",
                    "C:/Program Files (x86)/Arduino/hardware/tools/avr/lib/gcc/avr/4.9.2/include",
                    "C:/Program Files (x86)/Arduino/hardware/arduino/avr/variants/standard",
                ],
                "limitSymbolsToIncludedHeaders": false,
                "databaseFilename": ""
            }
        }
    ],
    "version": 3
}
```
Rough explanation:
- `"intelliSenseMode": "clang-x64"` is needed to play well with GCC
-  `__AVR_ATmega328P__` and `__AVR_ENHANCED__` normally only get defined during compilation, so they need to be explicitly called out for the Intellisense engine. **Note that these values will vary depending on the board.**
-  The `browse` and `includePath` settings are as specific as possible to guide the Intellisense engine to the correct definition for the board. e.g. `variants/standard` is specified instead of `variants` to get the correct pins_arduino.h for the Arduino Uno.

Ultimately, this extension should generate a lot of these configuration settings for the user, based on their platform and board selection.

Eternyt, 2018-02-18:
I'm still not able to get intellisens working on my Mac, anyone already gets it working?

aster94, 2018-02-18:
I think it is better to just wait, they added a milestone, so i think it would be fixed soon

derwentx, 2018-02-19:
@Eternyt have you tried @DavidKarlas ' solution? That worked for me

Eternyt, 2018-02-21:
@derwentx Yes, probably I did something wrongly...

czgtest, 2018-03-07:
@mundodisco8   have you got this issue solved?

mundodisco8, 2018-03-07:
Well, it really depends on what you mean by "solved".
My main problem was the myriad of squiggly lines that appeared. Now I know how to configure the intellisense file to solve it. For reference purposes, this is my ```c_cpp_properties.json```

In the browse section, add the folders where your code is located. ${workspaceroot} will cover your sketch's folder and subfolders. When a library is missing, intellisense will search in these folders recursively, you will see why this is useful later in this post. Add the folder where your libraries might be located. This depends on your configuration and the boar you're using:
```
"browse": {
                "path": [
                    "${workspaceRoot}",
                    "**path to your sketchbook**/libraries",      <==== Arduino default library location for users
                    "C:/Program Files (x86)/Arduino/libraries",  <====Same thing, but in your installation path
                    "C:/Program Files (x86)/Arduino/hardware",<====This two depend on what board you're using
                    "C:/Program Files (x86)/Arduino/avr"           <====I use a Uno on this example
                ],
                "limitSymbolsToIncludedHeaders": true
            }
```
On top of that, the IDE defines some words to enable or disable features, depending on your board. I don't know where to look at this, but in this thread people suggest some of these:
```
"defines": [
                "_DEBUG",
                "UNICODE",
                "ARDUINO=100",
                "__AVR_ATmega328P__",
                "__AVR__",
                "__AVR_ENHANCED__"
            ],
```
Also, some people mention that the intellisense mode should be changed to clang, don't know what the implications of this has.
```
"intelliSenseMode": "clang-x64",
```

After that, every time you include a library, go to the squiggly line, click on the lightbulb, and if the paths are right, it will suggest a location for that particular missing library. If your *browse* section is not set right, intellisense won't find the libraries. Choose the right path if some are suggested. In the case of that example, I know it's the first one.
[image]

When you do that, one entry is added to the *includepath* section of the c_cpp_properties.json file. Intellisense will keep searching within that recently added library, so you might need to add entries to other libraries included by your library, and so on. Once all the paths needed are added, your undefines will disappear.

BUT! I really don't see the point of having the extension installed just to have a bunch of buttons to upload the code and open the serial monitor (especially when loading the code implies recompiling from scratch and is so much slower than using the Arduino IDE). I think that the extension should add all the defines and libraries needed automagically based on the current board selected. I will edit the title of my issue to explain my actual issue now, as it has changed since I opened it.

fzoccara, 2018-03-18:
This contents on file "c_cpp_properties.json" works for me on mac, just replace <username> with your profile name:

> {
>     "configurations": [
>         {
>             "name": "Mac",
>             "includePath": [
>                 "/Applications/Arduino.app/Contents/Java/libraries",
>                 "/Users/<username>/Documents/Arduino/libraries"
>             ],
>             "browse": {
>                 "limitSymbolsToIncludedHeaders": false,
>                 "path": [
>                     "/Applications/Arduino.app/Contents/Java/libraries",
>                     "/Users/<username>/Documents/Arduino/libraries"
>                 ]
>             },
>             "intelliSenseMode": "clang-x64",
>             "macFrameworkPath": [
>                 "/System/Library/Frameworks",
>                 "/Library/Frameworks"
>             ]
>         }
>     ],
>     "version": 3
> }

Sneezry, 2018-03-20:
The suggested way is to use default engine and set `intelliSenseMode` to `clang-x64`.

`includePath` is for the default engine, and `browse` is for tag parser, so we should use `includePath` instead of `browse`.

However, `includePath` is not recursive currently. C/CPP extension team told us they will support recursive path in `includePath` 1 or 2 months later.

ArthurMa1978, 2018-04-09:
since ```browse``` also works, take this as a workaround and fix it temporally in next sprint

Sneezry, 2018-05-14:
We did some temporary workaround in v0.2.15, and we will keep seeking final solution

Pharap, 2018-07-04:
I believe it's worth mentioning that if `${config:arduino.path}` was allowed to be used in local user workspace files, that this:
```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "${workspaceRoot}",
                "${config:arduino.path}/hardware/tools/avr/avr/include",
				"${config:arduino.path}/hardware/arduino/avr/cores/arduino",
				"${config:arduino.path}/hardware/arduino/avr/libraries/EEPROM/src",
				"${config:arduino.path}/hardware/tools/avr/lib/gcc/avr/4.9.2/include",
				"${config:arduino.path}/hardware/tools/avr/lib/gcc/avr/4.9.2/include-fixed"
            ],
            "browse": {
                "limitSymbolsToIncludedHeaders": false,
                "path": [
                    "${workspaceRoot}"
                ]
            },
            "intelliSenseMode": "msvc-x64",
            "cppStandard": "c++11"
        }
    ],
    "version": 3
}
```

Would be a suitable solution for the base case (without factoring the board in).
Likewise the boards could be found in `"${config:arduino.path}/hardware/arduino/avr/variants/<board category>"` (which is only actually needed for `pins_arduino.h`).

Perhaps the same effect could be achieved programmatically?

I think adding local libraries probably counts as a separate issue perhaps?

sslupsky, 2018-09-21:
Hi @Sneezry There is a file named "includes.cache" that the Arduino build process creates in the build directory.  That file contains all the include paths for the project.  Could you use this file to fill the intellisense include paths automagically?

ljubomirb, 2018-10-15:
Installation did not work for me out of the box. I tried for several hours all sorts of combinations, and apparently, `defines `_may_ be the key, but I'm simply not sure what did I miss.
```

            "defines": [
                "ARDUINO_AVR_NANO",
                "F_CPU=16000000L",
                "ARDUINO_ARCH_AVR",
                "ARDUINO=10805",
                "__AVR_ATmega328P__"
            ],
```

sslupsky, 2018-10-24:
@ljubomirb Here is what I use for the MKRWAN1300:

            "defines": [
                "ARDUINO=100",
                "ARDUINO_ARCH_SAMD",
                "ARDUINO_SAMD_MKRWAN1300",
                "__SAMD21G18A__",
                "BOARD_ID_arduino_mkrwan1300",
                "_VARIANT_ARDUINO_MKRWAN1300_"
            ],

ljubomirb, 2018-10-24:
@sslupsky 
I'm not even sure what is or what was wrong. Most of my problems went away when I set **`tag parser`** as IntelliSense engine. So, something is missing either in "how to" instructions, or settings, or is otherwise making us doing the wrong steps. If "tag parser" should be used, then it should be set on automatically. For me, visual studio 201x community still works better with arduino plugin at the moment, but vscode, once fixed, definitely has a great future!

mundodisco8, 2018-10-24:
Tag parser should NOT be set. It's a fallback in case the normal mode
doesn't work. The reason why it "works" for you when you set it, it's
because you're telling intellisense to "ignore the errors" that we're
trying to fix here. So it's nothing that tag parser fixes the issue, tag
parser hides the issue.

danindiana, 2018-11-08:
Note to Linux VScode users: Make sure to use "Arduino: upload" as opposed to "upload using programmer." The latter will not upload to the arduino board but instead throw an "Exit with code=1" leaving you no details. 

Easy mistake to make since "upload using programmer" can appear as the first and only suggestion in the drop down menu unless you manually type Arduino: upload. 

OS: Ubuntu 16.04 xenial
 Kernel: x86_64 Linux 4.15.0-38-generic
 Uptime: 18m
 Packages: 3512
 Shell: bash 4.3.48
 Resolution: 2560x1080
 DE: Unity 7.4.5
 WM: Compiz
 WM Theme: Radiance
Radiance [GTK2]
, Radiance [GTK3]
 Icon Theme: ubuntu-mono-light
 Font: Ubuntu 11
 CPU: AMD Ryzen 5 1600X Six-Core @ 4GHz
 GPU: GeForce GTX 1050
 RAM: 2695MiB / 16034MiB

aroman, 2018-11-14:
If anyone else has been struggling to get proper Intellisense support in VS Code for Arduino, without randomly hunting pasting in defines and manually resolving pathing issues, I'd strongly recommend the PlatformIO VS Code extension.

After struggling to get proper code completion working with this plugin for a few hours, I installed that plugin and 5 minutes later everything worked perfectly. No config editing. It already does everything this bug (now a year old!) is seeking — it just sets everything up automagically. Even after hours of tweaking this plugin, I still was only able to get partial code completion by falling back to the tag-based engine, which severely limits the quality of code completion.

I have no affiliation with PlatformIO at all.  I wanted to love `vscode-arduino`, since it seemed lighter weight and I only care about Arduino — it's even officially recommended by Microsoft. But now I can definitely say that PlatformIO seems to be the way to go.

Wiqkszon, 2018-11-26:
[image]
[image]

Hi, I'm new on this program the problem I have is when I try to open an example (Blink) I got this 2 errors. I already try adding new paths on c_pp_properties.json, but it doesn't work. 
When I upload the code it just work fine but how do I get rid of those errors?

MarianMajik, 2019-01-01:
I've had problem with NodeMCU marking missing identifiers  - VS Code on OSX.

In my case helped **to force path for Arduino libraries**:
 
>             "includePath": [
>                 "/Users/mm/Library/Arduino15/packages/esp8266/tools/**",
>                 "/Users/mm/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/**"
>             ],
>             "forcedInclude": [  
>                  "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h"
>             ],
>            "macFrameworkPath": [
>                 "/System/Library/Frameworks",
>                 "/Library/Frameworks"
>             ],

clankill3r, 2019-01-14:
Yet another huge disappointment by Microsoft.
Hope this will get some love in the future.

Anyway, for `OSX` this is what I use at the moment for arduino 1.8.7. Be aware that I have a folder applications2 and that I renamed the arduino app to the version.

```
{
    "configurations": [
        {
            "name": "Mac",
            "includePath": [
                "${workspaceFolder}/**",
                "/Applications2/Arduino-1.8.7.app/Contents/Java/hardware/arduino/avr/cores/arduino",
                "/Applications2/Arduino-1.8.7.app/Contents/Java/hardware/arduino/avr/variants/standard",
                "/Applications2/Arduino-1.8.7.app/Contents/Java/hardware/tools/avr/include",
                "/Applications2/Arduino-1.8.7.app/Contents/Java/hardware/tools/avr/avr/include",
                "/Applications2/Arduino-1.8.7.app/Contents/Java/hardware/tools/avr/avr/include/avr",
                "/Applications2/Arduino-1.8.7.app/Contents/Java/hardware/tools/avr/lib/gcc/avr/5.4.0/include"
            ],
            "defines": [
                "ARDUINO=100",
                "_DEBUG",
                "UNICODE",
                "__AVR_ATmega328P__",
                "__AVR_ENHANCED__"
            ],
            "macFrameworkPath": [
                "/System/Library/Frameworks",
                "/Library/Frameworks"
            ],
            "compilerPath": "/usr/bin/clang",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "intelliSenseMode": "clang-x64"
        }
    ],
    "version": 4
}
```

sudoerChris, 2019-01-22:
In my experience, "compilerPath" might set to Visual Studio one, which cause a lots of false error, including the "missing includepath" warning mentioned above
change it to `"compilerPath": "<Your Arduino IDE installation path>\\hardware\\tools\\avr\\bin\\avr-gcc.exe"` solve the problem, no need to manually change includepath or add "browse" definition

TheTrigger, 2019-02-21:
this worked for me on Windows 10

```
"includePath": [
	"${workspaceFolder}/**",
	"C:/Program Files (x86)/Arduino/hardware/**",
	"C:/Program Files (x86)/Arduino/libraries/**",
	"C:/Users/Fabio/Documents/Arduino/libraries/**",
	"C:/Users/Fabio/Documents/Arduino/hardware/**"
],
```

elemenofi, 2019-03-01:
i had downloaded the Arduino IDE from arduino.cc using the Windows 10 "Get" feature... never again! i could never find my arduino directory and I almost go nuts. I get so stressed... Anyway I went back to arduino.cc and this time i downloaded a normal windows installer, put /Arduino in program files and then the snippets in this post made it work! hope that helps the next poor bastard!

oddstr13, 2019-03-07:
> In my experience, "compilerPath" might set to Visual Studio one, which cause a lots of false error, including the "missing includepath" warning mentioned above
> change it to `"compilerPath": "<Your [...]

This fixed the issue I was having with flash string macro (`F("Gimme more memory please")`) messing up the error reporting.

I also set `"intelliSenseMode": "gcc-x64"`, mostly just to match the compiler.

smamusa, 2019-04-20:
> 
> 
> For please this value to your vscode setting file to see whether fix the intellisense issue:
> 
> ```json
> "C_Cpp.intelliSenseEngine": "Tag Parser",
> ```

This is by far the simplest solution, I added this in my 
File -> Preferences -> Settings -> Extensions -> Arduino -> settings.json file
And it fixed the problems popping up.
NOTE: I'm using it only for navigating code, I'm still uploading through Arduino IDE.

hywhuangyuwei, 2019-05-10:
> If you meet something like `identifier "Serial" not defined`, also add "USBCON" into `defines` in `configurations`
> See [cpptools [...]

Thanks! It helps!

[65 more replies]

closed 2021-03-22 by adiazulay