microsoft/vscode-arduino #440: Support for upload through the USB volume

rei-vilo, 2017-11-06:
When plugging the board into the USB port, a USD drive or MSD appears.

How to upload the executable to the board by just copying it into the MSD?

This is how mbed operates. It is already implemented on the DevKit for upgrading the firmware.

czgtest, 2018-03-07:
@Sneezry  Can you help to support?

rei-vilo, 2018-03-07:
I'm using the following command on [embedXcode](https://embedxcode.weebly.com).

```
        UPLOADER             = cp
        TARGET_BIN_CP        = $(OBJDIR)/$(TARGET_NAME)_cp.bin
        COMMAND_PREPARE      = dd conv=notrunc bs=1 if=$(HARDWARE_PATH)/bootloader/boot.bin of=$(TARGET_BIN_CP) seek=$$((0x0000)) ;
        COMMAND_PREPARE     += dd conv=notrunc bs=1 if=$(TARGET_BIN) of=$(TARGET_BIN_CP) seek=$$((0xC000)) ;
```