microsoft/vscode-arduino #442: MQTT Library Conflict

rei-vilo, 2017-11-07:
If the `MQTT` library is already installed as a user's library under the sketchbook/libraries folder, *i.e.* 
`~/Documents/Arduino/libraries/MQTT`, compiling the `MQTTClient` example fails and shows the following error:

> In file included from ~/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.2.0/libraries/MQTT/examples/MQTTClient/MQTTClient.ino:2:0:
~/Documents/Arduino/libraries/MQTT/src/MQTTClient.h:5:20: fatal error: Client.h: No such file or directory
compilation terminated.
Multiple libraries were found for "MQTTClient.h"
 Used: ~/Documents/Projets/Arduino/libraries/MQTT
 Not used: ~/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.2.0/libraries/MQTT
 Not used: ~/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.2.0/libraries/MQTT
 Not used: ~/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.2.0/libraries/MQTT
 Not used: ~/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.2.0/libraries/MQTT
Using library WiFi at version 1.2.6 in folder: ~/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.2.0/libraries/WiFi 
Using library MQTT at version 2.2.0 in folder: ~/Documents/Arduino/libraries/MQTT 
exit status 1

The `~/Documents/Arduino/libraries/MQTT` as a user's library is required for all the other boards with Ethernet or WiFi capabilities.

andxu, 2017-11-23:
Please upgrade package to the latest version 1.2.1 and have a try.

dooriya, 2017-11-23:
Hi rei-vilo,

You can apply the following actions to resolve the MQTT Client library conflict:
- Rename scketch file MQTTClient.ino to **MQTTClientHello.ino**
- Rename the folder which contains the .ino file to MQTTClientHello.ino

[image]

And we will also update our MQTTClient exmaple in Production in next version to resolve this conflict with the general MQTT client for Arduino.

rei-vilo, 2017-12-18:
@andxu and @dooriya: Thank you for the solution, it worked.

closed 2018-02-09 by liydu