microsoft/vscode-arduino #443: Unable to debug Adafruit Feather M0

jdeane, 2017-11-07:
Hello,

I have a problem debugging an Adafruit Feather M0 using VSCode-Arduino. I have installed libusb to get Openocd to recognise the jlink.

I'm using:

- Segger Jlink with v6.20g firmware.
- Arduino extension v0.2.10 
- VSCode v1.17.2. 
- Windows 10 64 Bit

Here is the verbose output when I try to debug the application:

```
1: (140) LaunchOptions<LocalLaunchOptions xmlns='http://schemas.microsoft.com/vstudio/MDDDebuggerOptions/2014'
1: (149) LaunchOptions  ExePath='c:\Users\jdean\Desktop\Test\.build\adafruit_feather_m0\app.ino.elf'
1: (149) LaunchOptions  WorkingDirectory='c:\Users\jdean\Desktop\Test'
1: (149) LaunchOptions  TargetArchitecture='arm'
1: (149) LaunchOptions  ExeArguments=''
1: (149) LaunchOptions MIMode='gdb'
1: (149) LaunchOptions  MIDebuggerPath='C:\Users\jdean\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\bin\arm-none-eabi-gdb.exe'
1: (149) LaunchOptions  WaitDynamicLibLoad='false'
1: (149) LaunchOptions  DebugServer='C:\Users\jdean\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino\bin\openocd.exe'
1: (150) LaunchOptions  DebugServerArgs='-s C:\Users\jdean\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino\share\openocd\scripts\ -f interface/jlink.cfg -c "transport select swd" -f target/at91samdXX.cfg -c "telnet_port disabled" -c "tcl_port disabled"'
1: (150) LaunchOptions  ServerStarted='Info\ :\ [\w\d\.]*:\ hardware'
1: (150) LaunchOptions  FilterStderr='true'
1: (150) LaunchOptions>
1: (150) LaunchOptions    <CustomLaunchSetupCommands>
1: (150) LaunchOptions        <Command IgnoreFailures='false' Description=''>target remote localhost:3333</Command>
1: (150) LaunchOptions        <Command IgnoreFailures='false' Description=''>file c:/Users/jdean/Desktop/Test/.build/adafruit_feather_m0/app.ino.elf</Command>
1: (150) LaunchOptions        <Command IgnoreFailures='false' Description=''>load</Command>
1: (150) LaunchOptions        <Command IgnoreFailures='false' Description=''>monitor reset halt</Command>
1: (150) LaunchOptions        <Command IgnoreFailures='false' Description=''>monitor reset init</Command>
1: (150) LaunchOptions    </CustomLaunchSetupCommands>
1: (150) LaunchOptions    <LaunchCompleteCommand>exec-continue</LaunchCompleteCommand>
1: (150) LaunchOptions</LocalLaunchOptions>
1: (185) Starting: "C:\Users\jdean\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino\bin\openocd.exe" -s C:\Users\jdean\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino\share\openocd\scripts\ -f interface/jlink.cfg -c "transport select swd" -f target/at91samdXX.cfg -c "telnet_port disabled" -c "tcl_port disabled"
1: (282) Starting: "C:\Users\jdean\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\bin\arm-none-eabi-gdb.exe" --interpreter=mi
1: (291) DebuggerPid=21152
1: (322) ->=thread-group-added,id="i1"
1: (323) ->~"GNU gdb (GNU Tools for ARM Embedded Processors) 7.6.0.20140228-cvs\n"
1: (323) ->~"Copyright (C) 2013 Free Software Foundation, Inc.\n"
1: (323) ->~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law.  Type \"show copying\"\nand \"show warranty\" for details.\n"
1: (323) ->~"This GDB was configured as \"--host=i686-w64-mingw32 --target=arm-none-eabi\".\nFor bug reporting instructions, please see:\n"
1: (323) ->~"<http://www.gnu.org/software/gdb/bugs/>.\n"
1: (323) ->(gdb)
1: (328) <-1001-gdb-set target-async on
1: (332) ->1001^done
1: (332) ->(gdb)
1: (332) ->&"\n"
1: (332) ->^done
1: (332) ->(gdb)
1: (333) 1001: elapsed time 5
1: (337) <-1002-interpreter-exec console "set pagination off"
1: (343) ->=cmd-param-changed,param="pagination",value="off"
1: (343) ->1002^done
1: (343) ->(gdb)
1: (343) 1002: elapsed time 6
1: (343) ->&"\n"
1: (343) ->^done
1: (343) ->(gdb)
1: (343) <-1003-gdb-set solib-search-path c:\\Users\\jdean\\Desktop\\Test\\.build\\adafruit_feather_m0;
1: (353) ->1003^done
1: (353) ->(gdb)
1: (353) 1003: elapsed time 10
1: (353) ->&"\n"
1: (353) ->^done
1: (353) ->(gdb)
1: (355) <-1004-interpreter-exec console "target remote localhost:3333"
1: (379) ->~"Remote debugging using localhost:3333\n"
[... 169 lines cut ...]
openocd: Error: address + size wrapped(0xfffffffe, 0x00000004)
1: (2526) ->*stopped,reason="signal-received",signal-name="SIGINT",signal-meaning="Interrupt",frame={addr="0xfffffffe",func="??",args=[]},thread-id="1",stopped-threads="all"
1: (2526) <-1015-stack-list-frames 0 1000
openocd: Info : SWD IDCODE 0x0bc11477
openocd: Error: Failed to read memory at 0xfffffff4
openocd: Info : SWD IDCODE 0x0bc11477
openocd: Error: Failed to read memory at 0xffffffc0
1: (2547) ->1015^done,stack=[frame={level="0",addr="0xfffffffe",func="??"},frame={level="1",addr="0xfffffff9",func="<signal handler called>"},frame={level="2",addr="0x00000000",func="??"},frame={level="3",addr="0x00000000",func="??"}]
1: (2547) ->(gdb)
1: (2547) 1015: elapsed time 21
1: (2547) ->&"\n"
1: (2547) ->^done
1: (2547) ->(gdb)
Execute debugger commands using "-exec <command>", for example "-exec info registers" will list registers in use (when GDB is the debugger)
1: (2613) <-1016-stack-list-arguments 2 0 2
1: (2618) ->1016^done,stack-args=[frame={level="0",args=[]},frame={level="1",args=[]},frame={level="2",args=[]}]
1: (2618) ->(gdb)
1: (2618) 1016: elapsed time 5
1: (2618) ->&"\n"
1: (2618) ->^done
1: (2618) ->(gdb)

```

Any ideas?

andxu, 2017-11-23:
Seems your firmware has problems, see http://forum.segger.com/index.php?page=Thread&threadID=3392, to fix it, try to reset your firmware back to default, to make sure options at 00804000 are valid as the following descriptions:

"Good" board option bytes:

00804000 = FF C7 E0 D8 5D FC FF FF FF FF FF FF FF FF FF FF

"Bad" boards option bytes:

00804000 = FA C7 E0 D8 5D FC FF FF FF FF FF FF FF FF FF FF

liydu, 2018-02-09:
@jdeane , have you got this issue solved?

jdeane, 2018-02-09:
Hi,

I didn't try the solution above. I was able to get it working reasonably well by using the GDB from Atmel 7 and the Segger GDB server.

I'll try the solution above over the next few days and post back.

Thanks.

czgtest, 2018-03-07:
@jdeane  do you try the solution and fix work ?

jdeane, 2018-03-07:
Hi,

I tried the above fix, and unfortunately, I'm getting the same errors.

I have updated the J-Link software to 6.30f

I set the values starting at address 0x00804000 to FF C7 E0 D8 5D FC FF FF FF FF FF FF FF FF FF FF
I reloaded the bootloader using Atmel Studio 7, leaving erase option unchecked. I'm seeing these errors when debugging from VSCode:
```
Note: automatically using hardware breakpoints for read-only addresses.
1: (10826) ->^running
1: (10826) ->*running,thread-id="all"
1: (10826) ->(gdb)
1: (10826) ->&"\n"
1: (10826) ->^done
1: (10826) ->(gdb)
openocd: Error: at91samd.cpu -- clearing lockup after double fault
1: (10927) ->@"at91samd.cpu -- clearing lockup after double fault\n"
@"at91samd.cpu -- clearing lockup after double fault\n"
openocd: Polling target at91samd.cpu failed, trying to reexamine
openocd: Info : at91samd.cpu: hardware has 4 breakpoints, 2 watchpoints
openocd: Error: address + size wrapped(0xfffffffe, 0x00000004)
1: (10973) ->*stopped,reason="signal-received",signal-name="SIGINT",signal-meaning="Interrupt",frame={addr="0xfffffffe",func="??",args=[]},thread-id="1",stopped-threads="all"
1: (10973) <-1015-stack-list-frames 0 1000
openocd: Info : SWD IDCODE 0x0bc11477
openocd: Error: Failed to read memory at 0xfffffff4
openocd: Info : SWD IDCODE 0x0bc11477
openocd: Error: Failed to read memory at 0xffffffc0
1: (10995) ->1015^done,stack=[frame={level="0",addr="0xfffffffe",func="??"},frame={level="1",addr="0xfffffff9",func="<signal handler called>"},frame={level="2",addr="0x00000000",func="??"},frame={level="3",addr="0x00000000",func="??"}]
1: (10995) ->(gdb)
1: (10995) 1015: elapsed time 22
1: (10996) ->&"\n"
1: (10996) ->^done
1: (10996) ->(gdb)
Execute debugger commands using "-exec <command>", for example "-exec info registers" will list registers in use (when GDB is the debugger)
1: (11060) <-1016-stack-list-arguments 0 0 2
1: (11068) ->1016^done,stack-args=[frame={level="0",args=[]},frame={level="1",args=[]},frame={level="2",args=[]}]
1: (11068) ->(gdb)
1: (11068) 1016: elapsed time 8
1: (11068) ->&"\n"
1: (11068) ->^done
1: (11068) ->(gdb)
```

jdeane, 2018-03-07:
``

jdeane, 2018-03-07:
Apologies I closed the issue in error.

czgtest, 2018-03-16:
@Sneezry  Can you help out?