microsoft/vscode-arduino #444: How to add subboards in "usbmapping.json" file,

caniggia1, 2017-11-08:
Hi
I have added many boards to my "usbmapping.json" file, so i can debug them with st-link, and they work OK.
But i have problem to add boards to "usbmapping.json" if they are defined as subboards(in boards.txt) with diferent MCU family.
For example:
https://github.com/danieleff/STM32GENERIC/blob/master/STM32/boards.txt#L441
or
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/boards.txt#L78

How to distinguish between diferent subboards in "usbmapping.json" file,
so i can **set diferent "target" for diferent subboard**?
As far as i know the only way to distinguish between the boards is "id" in "usbmapping.json" file.
But in this example all the subboards have the same "id" (in this example "Nucleo_64").

Would be thankful for any help.

caniggia1, 2017-11-14:
Does no one know this?
Is it possible to set diferent targets for diferent subboards in "usbmapping.json" file?
Even if the answer is: "It's not possible".
I just wanna know if it's possible(and how) or not.

BR
caniggia

yaohaizh, 2017-11-15:
Yes, you can do this. You can set the different targets for different subboards as long as their vip & pid are different.

yaohaizh, 2017-11-21:
Close this as no response.

closed 2017-11-21 by yaohaizh

ChriD, 2018-06-05:
Same problem here. I have added 
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
as additional board manager.
 I can compile and upload sketches. But i can not debug the NUCLEO-F303RE eval board with the message

> "Debug for board Nucleo-64 is not supported by now."

I have added those lines to usbmapping.json
It seems Nucleo-64 subboards do have all the same pid?
How do i get the correct pid and vis. I used the values from this link
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/boards.txt#L78

```

 {    
    "index_file": 
    "https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json",
    "boards": [
      {
        "vid": "0483",
        "pid": "5711",
        "name": "Nucleo_64",
        "package": "STM32",
        "architecture": "STM32F3",
        "id": "NUCLEO_F303RE",
        "target": "stm32f3x.cfg"
      }
      ]
    },
```

and this to debuggerUsbMapping.json

```
 {
        "vid": "0483",
        "pid": ["374b","5711"],
        "name": "ST Link V2.1",
        "short_name": "stlink-2.1",
        "config_file": "stlink-v2-1.cfg"
    },

```

my arduino.json looks like this
```
{
    "board": "STM32:stm32:Nucleo_64",
    "configuration": "pnum=NUCLEO_F303RE,upload_method=MassStorage,xserial=third,usb=none,opt=osstd",
    "sketch": "Blink.ino"
}

```

How can i get this board debugged?
What am i missing?