microsoft/vscode-arduino #445: nothing happens for all command

MalcolmMielle, 2017-11-11:
I've installed this extension on my install and every time I try to verify a sketch, find the library or any other command provided by the extension, nothing happens. I can't even find an error message.

I don't know if that has to do with the fact the itellisense cannot see any library (which is a bug with multiple issues here)

Not sure what I can do to try to provide help in finding the cause. i've tried launche vscode in a terminal but still nothing is printed.

yaohaizh, 2017-11-13:
@MalcolmMielle Please provide more information for your issue. Version? It possible, list the snapshot for this issue.

MalcolmMielle, 2017-11-13:
No problem:
* visual-studio-code-1.18.0

How can I get the snapshot ?

Erutan409, 2017-11-23:
Yeah, I'm having the same issue.  What happened?  Seems to be affected after updating Visual Code after a couple months of absence.

yaohaizh, 2017-11-23:
@Erutan409 Do you install this extension? And what is your OS version? 
https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino

Erutan409, 2017-11-23:
@yaohaizh I don't know what happened, but I just kinda moved back and forth between options and it seemed to all of a sudden work.

Unfortunately another issue has been creeping up with me seeing `unknown protocol: c` 3 times in the bottom of the console.

Should this be happening on each startup of the Visual Code IDE, btw?

> [Starting] Update package index files...
> Looking for library 'user32'
> Adding paths from jna.library.path: null
> Trying user32.dll
> Found library 'user32' at user32.dll
> Looking for library 'shell32'
> Adding paths from jna.library.path: null
> Trying shell32.dll
> Found library 'shell32' at shell32.dll
> Looking for library 'Ole32'
> Adding paths from jna.library.path: null
> Trying Ole32.dll
> Found library 'Ole32' at Ole32.dll
> Loading configuration...
> Initializing packages...
> Preparing boards...
> Downloading platforms index... 
> Using proxy DIRECT
> Downloading platforms index... Downloaded 0kb of 198kb.
> Downloading platforms index... Downloaded 10kb of 198kb.
> Downloading platforms index... Downloaded 13kb of 198kb.
> Downloading platforms index... Downloaded 23kb of 198kb.
> Downloading platforms index... Downloaded 27kb of 198kb.
> Downloading platforms index... Downloaded 37kb of 198kb.
> Downloading platforms index... Downloaded 42kb of 198kb.
> Downloading platforms index... Downloaded 52kb of 198kb.
> Downloading platforms index... Downloaded 64kb of 198kb.
> Downloading platforms index... Downloaded 67kb of 198kb.
> Downloading platforms index... Downloaded 77kb of 198kb.
> Downloading platforms index... Downloaded 81kb of 198kb.
> Downloading platforms index... Downloaded 92kb of 198kb.
> Downloading platforms index... Downloaded 96kb of 198kb.
> Downloading platforms index... Downloaded 106kb of 198kb.
> Downloading platforms index... Downloaded 109kb of 198kb.
> Downloading platforms index... Downloaded 119kb of 198kb.
> Downloading platforms index... Downloaded 129kb of 198kb.
> Downloading platforms index... Downloaded 141kb of 198kb.
> Downloading platforms index... Downloaded 146kb of 198kb.
> Downloading platforms index... Downloaded 156kb of 198kb.
> Downloading platforms index... Downloaded 164kb of 198kb.
> Downloading platforms index... Downloaded 175kb of 198kb.
> Downloading platforms index... Downloaded 182kb of 198kb.
> Downloading platforms index... Downloaded 192kb of 198kb.
> Downloading platforms index... Downloaded 198kb of 198kb.
> Downloading platforms index... 
> Using proxy DIRECT
> Downloading platforms index... Downloaded 0kb of 0kb.
> Downloading platforms index... Downloaded 0kb of 0kb.
> unknown protocol: c
> unknown protocol: c
> unknown protocol: c
> Selected board is not available
> [Done] Updated package index files.
> [Starting] Update package index files...
> Looking for library 'user32'
> Adding paths from jna.library.path: null
> Trying user32.dll
> Found library 'user32' at user32.dll
> Looking for library 'shell32'
> Adding paths from jna.library.path: null
> Trying shell32.dll
> Found library 'shell32' at shell32.dll
> Looking for library 'Ole32'
> Adding paths from jna.library.path: null
[... 71 lines cut ...]
> Downloading platforms index... Downloaded 105kb of 198kb.
> Downloading platforms index... Downloaded 113kb of 198kb.
> Downloading platforms index... Downloaded 116kb of 198kb.
> Downloading platforms index... Downloaded 127kb of 198kb.
> Downloading platforms index... Downloaded 134kb of 198kb.
> Downloading platforms index... Downloaded 144kb of 198kb.
> Downloading platforms index... Downloaded 148kb of 198kb.
> Downloading platforms index... Downloaded 159kb of 198kb.
> Downloading platforms index... Downloaded 164kb of 198kb.
> Downloading platforms index... Downloaded 175kb of 198kb.
> Downloading platforms index... Downloaded 177kb of 198kb.
> Downloading platforms index... Downloaded 188kb of 198kb.
> Downloading platforms index... Downloaded 192kb of 198kb.
> Downloading platforms index... Downloaded 198kb of 198kb.
> Downloading platforms index... 
> Using proxy DIRECT
> Downloading platforms index... Downloaded 0kb of 0kb.
> Downloading platforms index... Downloaded 0kb of 0kb.
> unknown protocol: c
> unknown protocol: c
> unknown protocol: c
> Selected board is not available
> [Done] Updated package index files.

This is happening every time and I have no idea why.

czgtest, 2018-03-02:
update the latest vscode version and open an exist project to run command ,try again

czgtest, 2018-03-07:
@Erutan409 and @MalcolmMielle ,  Sometimes the URL's that board manager uses are down, and when this happens the board manager "removes" the already installed boards from the arduino installation.
then, VSCode Board Manager will automatic update,  That's why you can see logs about updating package and library.

this behavior is by design now,  we will plan to disable VSCode Board Manager automatic update next, and support users to customize whether this feature is automatically updated,  Is that help ?

czgtest, 2018-03-07:
please ref #394

Erutan409, 2018-03-09:
@czgtest Honestly, I wouldn't be a good reference for this, anymore.  I'm now using VisualMicro for Visual Studio Community Edition.

a-bertucci, 2018-03-16:
I had the same issue ("command arduino.* not found"), but resolved it by enabling the C/C++ extension.  It was a new installation of VSCode and the first thing I tried was the Arduino extension.  I'm guessing most users don't see this issue because C/C++ is probably enabled first.

macOS 10.13.3
Arduino 1.8.5
VSCode 1.21.1
Arduino extension 0.2.11

Hope this helps someone.

Sneezry, 2018-04-18:
No more feedback more than one week, close.

closed 2018-04-18 by Sneezry

ctastan, 2018-07-09:
just change proxy setting to "No proxy" via Arduino IDE. (Because Arduino Extention using Arduino IDE base settings)
File->preferences->Network->No proxy
Resolved for me.