microsoft/vscode-arduino #446: Adafruit Wiced Feather - Cannot find the OpenOCD from the launch.json debugServerPath property.Please input the right path of OpenOCD

karlrissland, 2017-11-12:
I am receiving the cannot find OpenOCD error when I try to debug a program on the Adafruit WICED Feather.

I have tried to run this on both OSX and windows without success.  Same error.

I basically followed the steps listed in the following issue, which was closed because the board being used wasn't specifically supported.
[https://github.com/Microsoft/vscode-arduino/issues/410](url)

However, the WICED is supported and I am running into the same issues.

Any tips?  Would love to get debugging working on both OSX and Windows....

Thank you....

yaohaizh, 2017-11-13:
@karlrissland It should be able to debug. But I don't get a copy of this board. 
1. What is the URL of your link. 
2. You can find some related information for your board from this link: 

http://www.stm32duino.com/viewtopic.php?f=41&t=2366

karlrissland, 2017-11-15:
Here is the URL to information about the board, I will take a look at the documentation you sent as well.  https://learn.adafruit.com/introducing-the-adafruit-wiced-feather-wifi.

czgtest, 2018-03-07:
I can't repro this issue, @Sneezry  ,Can you help to support?

czgtest, 2018-03-16:
@karlrissland  have you got this issue resolved?

Sneezry, 2018-04-18:
No more feedback more than one week, close.

closed 2018-04-18 by Sneezry

SunboX, 2018-10-04:
Same for me with Waveshare Driver Board 😢 (Node MCU 1.0 compatible) Works fine with Arduino IDE

leonbrag, 2020-06-03:
I have the same problem with Arduino Uno WiFi rev 2

skytrackerdave, 2020-06-15:
Same problem for me - Arduino Uno - using VSCode 1.46 with Arduino extension and Arduino version 1.8.12 installed

sytran, 2020-09-24:
Same problem for me - Arduino Nano Every - using VSCode 1.49.1 with Arduino Microsoft extension  v0.3.2.
I am able to "Verify" and able to "Upload". But when I execute "Run", I get the error.

"Cannot find the OpenOCD from the launch.json debugServerPath property.Please input the right path of OpenOCD"

I don't see any solution for either of the bug report #410 and #446 and yet they report as closed. I am missing something here? Could some one publish the working .json file that works on their system?

Here is the URL to information about the  Arduino Nano Every board.
https://store.arduino.cc/usa/nano-every

mrnams, 2021-09-15:
Same problem when i try to run simple led blink on ESP8266.