microsoft/vscode-arduino #449: Compiling is very slow

mladenbruck, 2017-11-23:
From moment pressing ctrl+Alt+R,  to the message "[Done] Finished verify sketch - Trusioma.ino"
pass 21 second.
21 second every time, even when I setup output folder in arduino.json
I tested it on 3 computers.

Same code in Arduino IDE  4 second...  
21 / 4 is 5.  So Arduino IDE is  times faster then vscode-arduino.

Is there a better reason NOT to use vscode-arduino ?

yaohaizh, 2017-11-24:
In the arduno.json file, you can set the output, then it will have the same performance as Arduino IDE. 

See the readme here: https://github.com/Microsoft/vscode-arduino

closed 2017-11-24 by yaohaizh

mladenbruck, 2017-11-24:
I'm not sure did I mentined that I setup output in arduino.json ???

> 21 second every time, even when I setup output folder in arduino.json
yes I did.

yaohaizh,  read the post carefully before answering..

sschicken, 2019-10-01:
So I've figured out why this is happening to me. If you use a NFS share on a linux machine, and you are building on a windows machine, the timestamps won't line up properly so the entire cache will be rebuilt every time you verify. To fix this, set your build directory to something local. It's annoying, usually other software that runs into this ends up using Fat File Times to decrease precision a touch and things will line up again, but until/unless that gets implemented the only solution I can see is to change to a local build directory