microsoft/vscode-arduino #450: Cannot set output folder to absolute path

iFreilicht, 2017-12-10:
So to me it seems sensible to but the build output folder into the %temp% directory (usually `C:\Users\<user>\AppData\Local\Temp`). This is how the Arduino IDE does it as well.

However, setting the output variable in `arduino.json` will lead to an error message when starting the build and upload:

###arduino.json

```
{
    "sketch": "bcm_data_direction.ino",
    "board": "arduino:avr:micro",
    "port": "COM4",
    "output": "C:\\Users\\<user>\\AppData\\Local\\Temp\\ArduinoBuild"
}
```
###Output

```
[Starting] Upload sketch - bcm_data_direction.ino
Looking for library 'user32'
Adding paths from jna.library.path: null
Trying user32.dll
Found library 'user32' at user32.dll
Looking for library 'shell32'
Adding paths from jna.library.path: null
Trying shell32.dll
Found library 'shell32' at shell32.dll
Looking for library 'Ole32'
Adding paths from jna.library.path: null
Trying Ole32.dll
Found library 'Ole32' at Ole32.dll
Loading configuration...
Initialising packages...
Preparing boards...
Verifying...
[Error] Exit with code=1
```

I also tried this with other absolute paths, and even on different drives. I get this very same error message every single time.

It does work perfectly when setting a relative path.

frgazor, 2017-12-16:
I can confirm this. There is some other funky behavior:  
"output": "../build" works fine, but "output": "build" will only compile my sketch for verify and won't compile for upload.

iFreilicht, 2017-12-18:
That is a different issue, though. The documentation specifically states that the build directory should not be inside the sketch directory as this can cause the build to fail.

In your example, `"output": "build"` is still a relative path, not an absolute one.

Peter-B-, 2018-02-05:
When I use an absolute path, I get a InvalidPathException:

When I the following line to my ```arduino.json```
```"output": "d:\\temp\\arduino"``` 

I see this error:
```java.nio.file.InvalidPathException: Illegal char <:> at index 53: d:\Dokuments\Arduino\generated_examples\VoiceRecord\d:\temp\arduino```

It looks like it is interpreted as a relative path. Is there a way to tell it, that this is a absolute path?

aster94, 2018-02-12:
+1 and also  #461

czgtest, 2018-04-03:
the version 0.2.12 support  set output folder to absolute path

closed 2018-04-03 by czgtest