microsoft/vscode-arduino #457: the Arduino Extension should do incrementally change to the additional URLs but not remove the existing links.

vsctest1, 2018-01-15:
Dev Kit installation packages will set custom Board Manger URL ,but this URL will be replaced when user also install  Arduino extension and set boards manager URL  in VS code.

Steps:
1. In the Arduino IDE, click File > Preferences.
2.In the Preferences dialog box, click the icon next to the Additional Boards Manager URLs text box.
3.In the pop-up window, enter the following URL, and then click OK.
https://raw.githubusercontent.com/VSChina/azureiotdevkit_tools/master/package_azureboard_index.json

4. Install Arduino extension in VS code.
5.In the VS code , click File > Preferences > Settings
6. set boards manager URL  as bellow

{
    "window.zoomLevel": 0,
    "arduino.additionalUrls": [
        "http://arduino.esp8266.com/stable/package_esp8266com_index.json",
        "https://adafruit.github.io/arduino-board-index/package_adafruit_index.json"
    ]    
}
7.save and close vscode
8. then wait about half an hour later 
1. In the Arduino IDE, click File > Preferences.>Additional Boards Manager URLs  
verify URL :  
http://arduino.esp8266.com/stable/package_esp8266com_index.json
https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
Issue :  the board URL for DevKit  was replaced.
Repro on : user install Arduino extension and set boards manager URL  in VScode.

liydu, 2018-02-09:
Ai...

czgtest, 2018-03-07:
this bug is open by me

czgtest, 2018-04-03:
the additional URLs of user settings will auto-update to preferences.txt file when run board manager command.

closed 2018-04-03 by czgtest