microsoft/vscode-arduino #459: ESP8266 NodeMCU 0.9 upload error

SakibFarhad, 2018-01-21:
When I verify it works well output: 
`[Starting] Verify sketch - esp8266LED.ino
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Archiving built core (caching) in: /tmp/arduino_cache_329972/core/core_esp8266_esp8266_nodemcu_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Serial,DebugLevel_None____,UploadSpeed_115200_177bf8d13a801b5e52a28f59133d464d.a
Sketch uses 275267 bytes (26%) of program storage space. Maximum is 1044464 bytes.
Global variables use 34932 bytes (42%) of dynamic memory, leaving 46988 bytes for local variables. Maximum is 81920 bytes.
[Done] Finished verify sketch - esp8266LED.ino`

[https://ibb.co/gc6uMw](url) <- every thing working good

but When I try to upload, It Fails:
`[Starting] Upload sketch - esp8266LED.ino
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Build options changed, rebuilding all
open preproc/ctags_target_for_gcc_minus_e.cpp: no such file or directory
[Error] Exit with code=1`

Some of the files get deleted:
[https://ibb.co/i4ZaTb](url)

closed 2018-01-24 by SakibFarhad

HontoNoRoger, 2018-10-24:
That problem happens every second verify run for me.
@SakibFarhad Did you manage to fix it? How did you do it?

Robotto, 2019-04-04:
This happens for me too: Ervy other verify results in this error.
Arduino version 1.8.3

Robotto, 2019-04-04:
Hm. I updated to Arduino Version 1.8.9, and the issue seems to have vapourised...