microsoft/vscode-arduino #461: add "output" to global user preferences

aster94, 2018-02-03:
i think it would be nice to have something like
`"arduino.output": "your_path"`
in the global settings accesible with `Ctrl + ,`

caniggia1, 2018-02-08:
You can add for example:
"output": ".build"
to ".vscode/arduino.json" file.

More on this page:
https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino

liydu, 2018-02-09:
Resolved.

closed 2018-02-09 by liydu

aster94, 2018-02-09:
@caniggia1  you misunderstand the question, or i didn't understand it
i know i can add the output in the LOCAL ".vscode/arduino.json" file, but i am suggestion this is not the best way to do it

I am suggesting that it would be better to have the possibility to choose once the built path in the GLOBAL  user preferences accessible with Ctrl + , 
if i didn't misunderstand please reopen the issue