microsoft/vscode-arduino #462: Error "Cannot read property 'map' of undefined" when calling Arduino commands

Karlo-Emilo, 2018-02-08:
I am using Windows 10.

Arduino initialize, Arduino Examples, Arduino board manager.. all gives the error:

Cannot read property 'map' of undefined

lcguida, 2018-02-08:
Same here. 

VSCode info:

````
Version 1.2.0
Shell 1.7.9
Renderer 58.0.3029.110
Node 7.9.0
Architecture x64
```` 
Arduino version 1.8.5

liydu, 2018-02-09:
@Karlo-Emilo can you paste the log details so that we can look into it? Thanks.

Karlo-Emilo, 2018-02-09:
The log:

messageService.ts:126 Cannot read property 'map' of undefined
e.doShow @ messageService.ts:126
e.show @ messageService.ts:105
e.resume @ messageService.ts:64
(anonymous) @ workbench.ts:1008
e.fire @ event.ts:142
(anonymous) @ quickOpenController.ts:534

bjepson, 2018-02-09:
I'm getting the same thing. I tried checking out the code and debugging the extension, but I think I'm just seeing front end errors and I don't know how to debug the back end where the error is probabl y originating. Here's the call stack; I don't see anything in the output or debug console though:

```(anonymous function) (c:\Users\bjeps\src\vscode-arduino\src\arduinoActivator.ts:19)
rejected (c:\Users\bjeps\src\vscode-arduino\out\src\arduinoActivator.js:7)
[ Promise.reject ] (Unknown Source:undefined)
rejected (c:\Users\bjeps\src\vscode-arduino\out\src\arduinoActivator.js:7)
[ Promise.reject ] (Unknown Source:undefined)
m (winjs.base.raw.js:1209)
enter (winjs.base.raw.js:852)
_run (winjs.base.raw.js:1068)
_completed (winjs.base.raw.js:1036)
e.resolveOk (lazyPromise.ts:65)
e._receiveReply (rpcProtocol.ts:135)
e._receiveOneMessage (rpcProtocol.ts:96)
(anonymous function) (rpcProtocol.ts:33)
(anonymous function) (rpcProtocol.ts:213)
(anonymous function) (event.ts:382)
e.fire (event.ts:142)
(anonymous function) (ipc.net.ts:81)
emitOne (events.js:96)
emit (events.js:191)
readableAddChunk (_stream_readable.js:178)
Readable.push (_stream_readable.js:136)
onread (net.js:560)```

dstaley, 2018-02-09:
Just wanted to note that I'm experiencing the same error (with the [same stack trace](https://github.com/Microsoft/vscode-arduino/issues/462#issuecomment-364378286)) running VS Code 1.20.0 on macOS 10.13.3 with Arduino IDE 1.8.5.

bjepson, 2018-02-10:
I reverted to VS Code 1.19.3, and the extension works fine. Looks like there were a lot of changes in 1.20, so it's likely related to one of those changes.

dstaley, 2018-02-11:
So it looks like this might be a VS Code bug, not related to this extension. To test, open a folder in VS Code (doesn't matter what kind of folder) and then try to run one of the Arduino commands. I can only replicate this issue when running in an empty workspace. The error seems to be coming from `vscode.workspace.findFiles` which seems to throw the error whenever the workspace is empty.

keyoke, 2018-02-15:
that's right I also just debugged this and basically looks like this error is thrown because the "arduino.json" is not found, also makes sense that it works then with an existing project.

keyoke, 2018-02-15:
I opened Issue here for this
https://github.com/Microsoft/vscode/issues/43757

I also have workaround here where if the findfiles promise is rejected I still initialize the Arduino settings 

https://github.com/keyoke/vscode-arduino/blob/184fb032cb3b0b20ba07a9818f793f0771a518b8/src/deviceContext.ts#L142

bleach103, 2018-02-26:
same problem I meet emmm.....
Windows 10
VScode version:1.20.1
Arduino IDE version:1.8.5
It's must be VScode‘s bug,just wait for fix.
For now just rolling back to 1.19

czgtest, 2018-03-07:
How to repro on:
1.	Open vs code new windows . here vs code not open any project.
2.	Press F1 and select Arduino :Examples command
3.	Verify the error: Cannot read property 'map' of undefined

angela-rph, 2018-03-08:
Same issue . Does anyone know how to fix this problem ?

Sneezry, 2018-03-08:
We just released 0.2.11 which should have fixed this issue, please upgrade to 0.2.11 and have a try.

czgtest, 2018-03-09:
@Karlo-Emilo @angela-rph ,  can you upgrade the Arduino extension to version 0.2.11  and have a try again ?

[image]

angela-rph, 2018-03-09:
Does not work as expected for me. Verification and Upload work fine but I got some error messages now  : 

[image]

keyoke, 2018-03-10:
Switching C_Cpp.intelliSenseEngine to Tag Parser in user settings as described in these instructions fixed it for me - https://stackoverflow.com/questions/47401731/issues-setting-up-mxchip-vscode-arduino-mx3166-environment-on-mac/47448652#47448652

lirenhe, 2018-03-15:
The original problem was solved. Close it.

closed 2018-03-15 by lirenhe