microsoft/vscode-arduino #463: Feature request: Enable input for serial monitor

liydu, 2018-02-09:
Ref: #337, #409, #448.

czgtest, 2018-03-06:
Now we don't have UI to support  input for serial monitor, just have a command to input text to serial port
[image]

aster94, 2018-03-07:
Well, better then nothing, thank for the great work
Hope to see the UI soon or later

czgtest, 2018-05-11:
the design of Serial Monitor is different from Arduino IDE,  and we support the same behavor in serial monitor. maybe user don't know how to use serial monitor in extension, here are some steps:

steps
1. run command : Arduion:upload
2. click "open Serial Monitor" button to open the serial port
[image]
3. select "Band Rate" and "Serial Port Line Ending" as you want
[image]
4. press F1 and select command : Arduino:send text to serial port

closed 2018-05-11 by czgtest

davelnewton, 2019-08-23:
This should probably be in the docs somewhere (and maybe it is); took me 10 minutes to find this ticket :/

ahorvat, 2020-01-05:
> This should probably be in the docs somewhere (and maybe it is); took me 10 minutes to find this ticket :/

Agreed, documentation can use some improvement

elektronikworkshop, 2020-01-23:
Having a good input for the serial monitor is quite important for interactive software. Most of my projects include a serial command line interface. The "Send Text to Serial Port" is pretty cumbersome when having to configure devices that way.

Using an external terminal like minicom or similar is inconvenient too because any other software will interfere with uploading.

Perhaps you can recycle the code of the debug terminal which has an input field - this would at least save some coding.

I would chip in some money too if that would help.

Thanks for the pretty cool extension so far!

Johnlon, 2020-02-08:
+1

Using "send to serial" is to painful to use.
See Arduino IDE - when focus in on serial monitor then typing is sent to the serial comms.
Need same or better

elektronikworkshop, 2020-02-08:
HI Johnlon,

I'm currently writing a better IntelliSense-backend for vscode-arduino, because the IntelliSense implementation was even more painful than the serial monitor and no one cared of this issue even if the reports date back before 2017.

I'm now pretty familiar with the code base. As soon as I've finished this rather extensive task I'll probably address this issue and will try to enhance it's functionality compared to Arduino's IDE (command history is one of the things I thought of).

You can track the project's progress [here](https://github.com/elektronikworkshop/vscode-arduino/blob/intellisense-autoconfig/BRANCHNOTES.md).

Regards
EW

Johnlon, 2020-02-09:
Appreciated.
I've looked at lots and f options recently and honestly the best I've come up with is editing in vscode hitting save and then running in Arduino to use the serial monitor. The Arduino software However doesn't see the vscode file updates as it is such a primitive tool so I have to restart it frequently.
Can't believe this is the current state of play with Arduino tooling.
I tried eclipse plugin and it was impossible to use. And even though I  use intellij for the day job I ouldn't make it work for the Arduino. Will try clion instead.

This seems a problem crying out to be solved.

Johnlon, 2020-02-09:
I might actually try and patch the Arduino software to load file updates if I can find the time

Johnlon, 2020-02-09:
Ps found this....
https://forum.arduino.cc/index.php?topic=386171.0

Johnlon, 2020-02-09:
There are actually three things I need to make vscode work well for this

1) allow input on the serial monitor in a convenient way - ie just type and hit return, just like the Arduino IDE
2) have the serial monitor window NOT keep turning off autoscroll (there is a separate ticket for this)
3) have the option of the serial monitor and/or compile window auto clear each time the sketch is compiled

Plus there is the annoying default where the compile runs in verbose mode and we have to manually edit config files to turn off the trace output

elektronikworkshop, 2020-02-09:
Thanks for your input - exactly: These are the features I'm looking for as well. I added them to the list of future work in my branch notes. We can only hope that I'll get that merged later on. If not we could detach my fork and publish it as new extension. Latter wouldn't be the preferred way but it has its advantages: 

* Being free means that fixes and improvements can be implemented much quicker.
* The existing user base will follow - the complaints about the bad IntelliSense implementation are all over the place. And the commit activities here are rather modest.
* I have my problems with the Microsoft repository since you have to sign a [CLA](https://github.com/microsoft/vscode-arduino/pull/963) to get pull requests approved. You loose any control of your contributed code - this keeps some really cool people from contributing most probably

Johnlon, 2020-02-09:
Oh and one more.
I want the serial output and perhaps compile windows to be undocked or at least I want them to sit to the right of my code window but they seem rigidly stuck at the bottom of the screen.

Johnlon, 2020-02-09:
And I would probably prioritize ease of use over better editing/intelligence.
Hope that helps.

elektronikworkshop, 2020-02-09:
I incorporated all into my [notes](https://github.com/elektronikworkshop/vscode-arduino/blob/intellisense-autoconfig/BRANCHNOTES.md)
I try to remember this post when I get to its implementation ...

elektronikworkshop, 2020-02-10:
I just installed vscode 1.42 and what did I see in the release notes:

>> Panel placement - Move the panel (Output, Terminal) to the left, right, or below the editor.

### Moving the panel
The panel, which holds the Output, Debug Console, Terminal, and Problems panes, can now be moved to the left side of the editor.

The command View: Toggle Panel Position (`workbench.action.togglePanelPosition`) was removed in favor of the following new commands:

View: Move Panel Left (`workbench.action.positionPanelLeft`)
View: Move Panel Right (`workbench.action.positionPanelRight`)
View: Move Panel To Bottom (`workbench.action.positionPanelBottom`)

----
One thing less on my ToDo list...

marcmerlin, 2020-05-17:
So, this bug is 2 years old, it's closed, but really the current implementation is not great as has already been said
In my workflow, I need to send 'n' a bunch of times (maybe 20 or 50). 
right now, it's
F1
select send character
type character + newline

Is there anything easier than that possible right now? it's quite cumbersome if you have to repeat a lot of times. Thanks.

piratebriggs, 2020-08-03:
The PlatformIO vscode extension has a workable serial monitor implementation - I've not checked that project's licence however.

marcmerlin, 2020-08-04:
Would it be acceptable to re-open this issue until the serial terminal is a bit more terminal like?

RicardoRickTron, 2020-09-02:
> +1
> 
> Using "send to serial" is to painful to use.
> See Arduino IDE - when focus in on serial monitor then typing is sent to the serial comms.
> Need same or better
 it's is really painful and ridiculus.

clouds56, 2021-06-01:
Is it possible to make it looks a true TTY?
That is to say we could typing in the output window, just like the terminal.

reggiiie, 2022-05-19:
any updates on this or possible workarounds? it has been 4 years

avtalion, 2022-06-21:
> the design of Serial Monitor is different from Arduino IDE, and we support the same behavor in serial monitor. maybe user don't know how to use serial monitor in extension, here are some steps: [...]
the line ending menu next to <select programmer> does not exist on my device. do i need to enable it somehow? text automatically receives a line ending, and i would like it to stop.

bklanders, 2022-07-04:
It would be awesome if we could revisit this. not having some kind of panel to be able to easily and repeatedly enter Serial inputs makes it so I can't realistically use solely vscode as my arduino build tool of choice (despite the awesome intelisense and coloration)

F1 > Down Arrow > Enter > Input Command > Enter is just too many steps :-(.

If I had a way of keeping that command window open, that would be enough.